luc-github/ESP3D #510: Upload file "Error 0 :

reynolds99i5, 2020-09-25:
Upload file "Error 0 :"

Board SKR 1.4Turbo on LPC1769 Chip
Compiled ESP3D 2.1 and ESP3D 3.0 and both have this same error while uploading file.
Board software Marlin Marlin-bugfix-2.0.x
I have serial comunication, i see SD card files but this same problem "Error 0 :".
After this exception my bed and nozzle temperature is set to printing temperatures (60 and 200).
After error when i refresh sd files (from WEB-GUI) i see this:

[image]

I have definied in marlin configuration.h:

#define SERIAL_PORT 3
#define LPC_PINCFG_UART3_P4_28

This means:

For UART3:
define LPC_PINCFG_UART3_P4_28 to use pin 82 and 85, PORT 4-28/29, FUNC 3

M20 code output:
M20
Begin file list
CAT-351.GCO 34432620
TEST_B1.GCO 32784
DOG-251.GCO 12449184
PIG-4H1.GCO 19126652
SPIRAL1.GCO 241643307
CE3PRO2.GCO 58462550
11_BAZ1.GCO 8411667
12_MOC1.GCO 0
13_CHL1.GCO 7516263
14_BLT1.GCO 4559645
15_WYP1.GCO 3570906
TEST_A1.GCO 5605723
PLA12_~1.GCO 3406518
PLA11_~1.GCO 7844806
PLA13_1.GCO 7356504
PLA14_1.GCO 3986674
CE3PRO1.GCO 508092
AGA1.GCO 6170297
PLA14_2.GCO 863132
OSA1.GCO 13528471
End file list

M115
FIRMWARE_NAME:Marlin bugfix-2.0.x (Sep 18 2020 17:56:06) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 Pro EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:0
Cap:RUNOUT:0
Cap:Z_PROBE:0
Cap:LEVELING_DATA:0
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0

luc-github, 2020-09-25:
what are esp3d fw and web ui version?

luc-github, 2020-09-25:
I see : Cap:AUTOREPORT_TEMP:1
you need to be sure autoreport temperature is not ongoing when doing upload - it may disturbe communication

luc-github, 2020-09-28:
@reynolds99i5 any update ?

reynolds99i5, 2020-09-28:
When i sent M155 S0 in M115 still is Cap:AUTOREPORT_TEMP:1.
But when i send command M155 S0 temperature readings stopped.
So i retry upload file and have this same result.

Bellow screenshot from console 
[image]

luc-github, 2020-09-28:
please answer previous question:
>what are 2.1 esp3d fw and web ui version?

reynolds99i5, 2020-09-28:
UI: 2.1b20 / FW: 2.1.0.c26

reynolds99i5, 2020-09-28:
> what are 2.1 esp3d fw and web ui version?

UI: 2.1b20 / FW: 2.1.0.c26

luc-github, 2020-09-28:
wow this are super old version, please update to latest git versions FW 2.1.1.b8 and UI 2.1b66

luc-github, 2020-10-05:
I guess issue is solved

closed 2020-10-05 by luc-github