luc-github/ESP3D #511: [Question]<how to upload marlin and esp3d in one modul esp32>

liwajaya, 2020-09-30:
how do i combine marlin and esp3d in one esp32 module i will use esp32 like have [Maplerain](https://github.com/maplerainresearch/MRR_ESPA) or grbl have
[Bdring](https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Machines/3axis_v4.h) without additional board.  I am not clear in the order of how to upload ... is Marlin uploaded first, then uploads esp3d like that?  or vice versa.  I hope anyone can help

luc-github, 2020-09-30:
Marlin instaled in ESP32 board use ESP3DLib you need to configure it in Configuration_adv.h

https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/Configuration_adv.h#L3391-L3410

luc-github, 2020-10-05:
I think issue can be closed now

closed 2020-10-05 by luc-github