luc-github/ESP3D #512: [BUG]Some Wiki images are broken

luc-github, 2020-10-01:
Some of the wiki images links are broken - I am guessing it is due to main branch is now changed 

I will work on new wiki solution to avoid such issue 
- use a special branch called wiki to store all pictures in one location vs move them to issue tracker 
- use github-action to sync a wiki repository of wiki branch to handle wiki

luc-github, 2020-10-01:
some external links are also broken 
[image]

and I am wondering if any images in issue tracker could also be lost in some way if github do some cleaning

luc-github, 2020-10-01:
doing a wiki branch with all data wiki page + images will secure this
from another point of view doing PR process will loose some flexibility and ease of use of editing WiKi

Any comment on this ?

DivingDuck, 2020-10-01:
It's always the same conflict between active development and documentation of a released version. There is no best way I think. Maybe having a development wiki separated from a relased version is a good choice. Then you have something to use as documentation for the development (like you do since quite a while as an "issue document") and when a function/feature is ready for a "productive release version" move this part of documentation to the productive wiki version. This can help to kind of develop a documentation in an early phase. Updating it further during the implementation phase like a workbook is so much easier and result in a semi finished document that is already available and need maybe only a last fine tuning.

luc-github, 2020-10-02:
I am not good at documentation and to be honest not really willing to spend time to improve because most of people do not read it ... 
On another project I have a simple read me for installation  - always up to date with code, because installation need strict actions to success or you have very specific error that is also described in read me and in FAQ
=> 90% people do not even check read me / FAQ but follow outdated video on youtube and come back to github complaining  it is not working due to the installation error.

That is why I was thinking for ESP3D 3.0 to use youtube videos instead of writing docs / doing screenshots etc.... but this is another topic

In my idea, the current wiki is more for people looking for the setup they need, because too many printer boards and configurations right now. so people who want to share their working setup, can help new comers wondering if it will working on their configuration, and how to succeed. I cannot do this because I do not have all hardware that is why this is comunity effort which was good until now IMHO

I want to be sure information is not lost due to broken links and community able to update it - but github wiki is lack of file upload feature which lead to several files locations, leading to current situation of broken links.

I think I will go to github-action to sync wiki repository - we will loose some edition ease of use but will be sure information is always there and as wiki will follow the production branch it will be always available for `futur` old versions also

luc-github, 2020-10-04:
@crispycritter911 do you still have the image you uploaded in https://github.com/luc-github/ESP3D/wiki/Wanhao-D6---MonoPrice-Maker-Ultimate ?
your link is dead - please prove the image here or do a PR to update the wiki - thank you

luc-github, 2020-10-04:
@jayjojayson do you mind if I move the 2 images for https://github.com/luc-github/ESP3D/wiki/D1-mini from your repo to wiki directory ? thank you

luc-github, 2020-10-04:
wiki is now sync with wiki directory, links should be fixed - just waiting for @crispycritter911  and @jayjojayson feedback to close topic

if I missed any link feel free to raise up 😅

luc-github, 2020-11-02:
if no update I close topic - feel free to raise / PR missing images

closed 2020-11-02 by luc-github