luc-github/ESP3D #514: Adding support for addressable led strips - Where to start?

fermino, 2020-10-09:
I skipped the questions for the feature request as I'm not requesting it but rather trying to implement it myself. I'll make a pull request when it's finished :)

I want to add support for addressable led strips, at least for the WS2812, but before to start coding I wanted to ask a couple questions. Those LEDs can be controlled by using only one data pin for the whole strip, so it could work even on the ESP-01 boards (using GPIO2 as output and leaving GPIO0 available to change the boot mode if needed). 

I saw there are ESP3D specific commands (ESPxxx), what would be the best place to start adding support for an extra one?
There's a lot already, any suggestion about the numbering? I think for best compatibility the format should correspond to the M150 command. 
Also, I was thinking about enabling some sort of "interception" feature that would catch the M150 requests and control de LEDs accordingly. This would be good for some printers that don't have available GPIO pins (which is actually my case). 

An option would be to use FastLED, but I'm guessing there might not be enough flash space. Anyways, I'll try it out and see what happens. 

With all this, before I start digging into the code, what would be a good place to start? Any ideas or suggestions?

Anyways, thanks for this awesome project, it's really useful!
-fermino-

luc-github, 2020-10-10:
Well numbering will depend of action but it think will be in 2xx like other actions commands

What actions + parameters do plan to implement ? it would help to understand your target as there are several ways you can control led strip:  scripted, automatic scenari, reacting actions, etc...

About "interception" this should be avoid, especially on ESP01 : need to check FW is Marlin then if command is M150 - this add some code depending FW I try to remove from ESP3D, you will implement already an ESP command so it should be ok but it will also depend what user cases you are thinking about.

Alsobe noted ESP3D  2.1.x is now feature frozen to be stable, only bug fixes are implemented, new feature go to 3.0 which allows easily any new feature in code structure / configuration

So IMHO  I would create a device module like `sensor`.

You mention FastLED library but is it so big ? I see it is compatible EP8266/ESP32 it may be easier to use a robust multi device support library than try to do a stripped one, even if first only one device is supported - this is what I did with sensor but you are free to choose your solution.

feel free to join discord channel if you have questions, I am looking foward to see the result ^_^

luc-github, 2020-10-12:
@fermino did  this answered your question ? can we close ticket ?

fermino, 2020-10-24:
@luc-github sorry for the long delay! I remember writing a response to this but I guess I didn't send it!
I had a problem with my 3D printer so I put this stuff aside for a couple of weeks (until the new inductive sensor arrived!)

Yes, for sure, I'll work on version 3 and will make a pull request as soon as I have it working!
What is the ETA for V3?

You can close the ticket if you want, anyways if I have any questions I'll post it here :)

Thank you!

fermino, 2020-10-24:
@luc-github so, the plan is to implement something similar to Marlin's M150 command. 

The usage, as taken from Marlin's reference is as follows: 

```
M150 [B<intensity>] [I<index>] [P<intensity>] [R<intensity>] [U<intensity>] [W<intensity>]

[B<intensity>] | Blue component from 0 to 255
[I<index>]     | Index from 0 to n (Requires NEOPIXEL_LED)
[P<intensity>] | Brightness from 0 to 255 (Requires NEOPIXEL_LED)
[R<intensity>] | Red component from 0 to 255
[U<intensity>] | Green component from 0 to 255
[W<intensity>] | White component from 0 to 255 (RGBW_LED or NEOPIXEL_LED only)
```
It lets you set each pixel independently or set a color for the whole strip. 

A was thinking about `ESP150`, as it would be pretty recognizable, but I saw it's already taken. I saw also that most of the subsets of numbers are taken too, so, is there any command number that you think is more reasonable for this LED use?

Implementing this should be pretty straightforward, so I should have it working quickly :)

Thank you!

luc-github, 2020-10-24:
3.0 fw is in alpha - I am now still working on esp32Cam integration - I was also busy these days
3.0 webUI is also on alpha working on camera integration as well and also need to do wizard 
so ETA is when is ready sorry

For output the 200 range is what I have used, so let say ESP260

closed 2020-10-24 by luc-github