luc-github/ESP3D #515: [Documentation] Fatal Exception (28) error

gordo3di, 2020-10-14:
I am using the ESP01 module from biqu, ESP3D 2.1 firmware and a USB to ESP adapter from Amazon. I followed this tutorial:

https://github.com/luc-github/ESP3D/wiki/Install-Instructions

After several days of pulling my hair out I just wanted to document how I fixed my problem.

I followed the instructions above and it got me 99% of the way through but every time I flashed the module I would get an endless stream of Fatal Exception (28) messages in the serial monitor. This happened even with example esp8266 sketches.

The only way to fix it was to flashing using DOUT mode instead of DIO. Took me forever to find this forum post:
https://www.esp8266.com/viewtopic.php?p=84897

Again, just putting it here to help others.

luc-github, 2020-10-14:
dio,  dout , qio,  the setting depend of board, indeed dout is the compatibily mode now,  I may change platformio.ini accordingly
thank you for raising it

luc-github, 2020-10-30:
I have updated 2.1 and 3.0 platformIO.ini with dout

Thank you

closed 2020-10-30 by luc-github