luc-github/ESP3D #516: [BUG]esp8266 2.1.1 esp3d and ftf35 BIGTREE TFT35 V3.0 E3.26.x

islonina, 2020-10-14:
On AP setup i dont see wifi when connected to TFT35(correct pinout)
Working wifi when connecting only CHEN and power

**Expected behavior**

**Screenshots**
If applicable, add screenshots to help explain your problem.

ESP3D Firmware:
- ESP core version: [2.1.1]
- Version:[e.g 2.1b4]
- Web UI Version:[e.g 2.1b14]   
- Flash method:
Arduino
Platform IO
ESP8266 tool

- Any change done in FW
  No changes

Target Firmware:
 - Name: Marlin 2.0.7
- TFT35 E3.26.x 
 

Board used (please complete the following information):
 ESP8266 ESP-01
 Flash size: 1M 512kb

islonina, 2020-10-14:
> Thank your for submiting, please be sure you followed template or your issue may be dismissed. if you deleted the template it is [here](https://github.com/luc-github/ESP3D/blob/2.1.x/.github/ISSUE_TEMPLATE/bug-report---problem.md)

I added all information firmware on screem, esp, program useed to flash

luc-github, 2020-10-15:
this is an hardware issue, I am guessing you are not using an ESP01 provided by BTT but another one.

I got same issue with my old ESP01 boards, and the answer is in your description actually.

On BTT ESP01 (or may be all new models?) the CH_PD is already connected to power unlike old ones , so TFT do not bother to do the connection

So you need to do a bridge to it by yourself 
Here what I did on some of my boards (sorry not clear image but readable enough):

[image]

islonina, 2020-10-15:
Thx. Simple solution. Now everything works without a problem

[image]

luc-github, 2020-10-15:
great - have funt

closed 2020-10-15 by luc-github