luc-github/ESP3D #517: [BUG]ESP3D can't control ender3 but usb can

a491113058, 2020-10-21:
**Describe the bug**
I use the ESPA motherboard. Now I can open a web page or upload files to an SD card, but I can't control the printer.
The printer can be linked and controlled through serial port.
So what's wrong with the hardware? I don't know what went wrong

**Screenshots**
[image]

**ESP3D Firmware:**
[ESP420]plain
Chip ID: 57576
CPU Frequency: 240Mhz
CPU Temperature: 76.7C
Free memory: 161.14 KB
SDK: v3.2.3-14-gd3e562907
Flash Size: 4.00 MB
Available Size for update: 1.25 MB
Available Size for SPIFFS: 1.31 MB
Baud rate: 115200
Sleep mode: None
Web port: 80
Hostname: marlinesp
Active Mode: STA (AC:67:B2:D9:E8:E0)
Connected to: PDCN
Signal: 100%
Phy Mode: 11n
Channel: 1
IP Mode: DHCP
IP: 192.168.99.22
Gateway: 192.168.99.1
Mask: 255.255.255.0
DNS: 192.168.99.1
Disabled Mode: AP (AC:67:B2:D9:E8:E1)
FW version: Marlin 2.0.7 (ESP3D:1.0.8)

**Target Firmware:**
FIRMWARE_NAME :  Marlin 2.0.7
 
 tip: to get firmware info type `M115` in console    

**Board used (please complete the following information):**
 - MCU: ESP32
 - Name:ESPA V1.2

**Additional context**
Add any other context about the problem here.
Wiring, decoded stack, etc...

luc-github, 2020-10-21:
I am guessing it is Marlin configuration problem - they changed settings recently
https://github.com/MarlinFirmware/Marlin/pull/19812#issuecomment-713080930

enable https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/Configuration.h#L112 
and it should communicate

exilaus, 2020-10-21:
[image]
just for explain clean configure serial port as in picture and websoket be initialized.

a491113058, 2020-10-21:
This is indeed the reason, and it can work. Thank you very much!
But I encountered a new problem, the limit switch of the X axis is closed, which will cause the network service to restart.

luc-github, 2020-10-21:
what do you mean ?

a491113058, 2020-10-21:
X limit switch is triggered == will cause the network service to restart.

Please see the picture:
[image]

luc-github, 2020-10-21:
you have ESPA so did you applied the patch I suggested to vivian-ng ? 
https://github.com/MarlinFirmware/Marlin/pull/19812#issuecomment-713204065

luc-github, 2020-10-21:
here the path of the change need to be done https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/HAL/ESP32/HAL.cpp#L89
you can just delete `i2s_init();` in your case - it is more easy than type ;)

exilaus, 2020-10-21:
Luc placing in pull this fix too for esp32.  today at office little difficut to manage that but surely in future bugfix can remove that issue for no i2s steppers board

luc-github, 2020-10-21:
@exilaus yes thank you

luc-github, 2020-10-24:
@a491113058 fix has been merged in bug-fix already - does it fix your issue ?

luc-github, 2020-10-30:
I guess no more issue

closed 2020-10-30 by luc-github