luc-github/ESP3D #518: WiFi stays in “Disconnected” state

Felix-ArduPilot, 2020-10-23:
Hi!

I’m new to ESP3D but I thought it would be very useful to be able to connect to my printer via WiFi. So I followed the installation instructions very carefully, but unfortunately I still can’t connect to the WiFi module.

My printer is a JGAurora A5. It has the MKS. Gen_L V1.0 board (Marlin firmware) and the MKS TFT28 V4.0 touch screen display. I connected the MKS WIFI TFT v1.0 module to the display. It has a connector that fits into the display, so I just plugged it in.

Then I flashed the firmware with the SD-Card-method. I tried MKS WiFi and also ESP3D. Both with the same result: In STA mode The WiFi remains disconnected. In AP mode, I can see the WiFi and connect to it and open up the web interface. Surprisingly the display of the printer still says “Disconnected”. With the MKS WiFi firmware it also changes to “Exception” sometimes. ESP3D says disconnedted the whole time.

I can upload new firmware and make changes in the menu of the webinterface, but I have no control over the printer. There is just no reaction.

Any ideas what the problem could be?

luc-github, 2020-10-23:
sorry you mean what ever fw you use MKS or ESP3D you cannot connect to wifi 
or it is just the tft do not detect wifi?

Felix-ArduPilot, 2020-10-23:
I can see the WiFi with my PC and I can connect to it and open the webinterface, but I can’t control the printer and the tft display always says “Disconnected”. It never changed to “Connected”. I have the same behavior with MKS firmware and ESP3D.

luc-github, 2020-10-23:
you know that you need to configure your tft with and confiruration file to enable what kind of wifi module you use ?

I do not use mks FW anymore - I use BTT version on my mks TFT now so I do not remember where is the file

Felix-ArduPilot, 2020-10-23:
You mean the mks_config.txt file? I’ve set the parameter inside that file like the following:

#wifi model(MKS WIFI TFT:0;  MKS HLK-WIFI:1)
>cfg_wifi_type:0

luc-github, 2020-10-23:
for MKS TFT WIFI and MKS FW I cannot comment what the pb is 
For ESP3D must set MKS HLK-WIFI:1 because it is the emulated protocol - I could not understood the MKS TFT WIFI protocole, also pinout is different between both protocole so need to use dupond cable to connect MKS TFT WIFI module to tft 

Using BTT FW it is supported natively so I did not spent time on this

Felix-ArduPilot, 2020-10-23:
Ah okay, thank you. That is a good tip!
So I have to set MKS HLK-WIFI:1 and then I don’t need a dupond cable, is that correct?

luc-github, 2020-10-23:
you need - pinout is different : https://github.com/luc-github/ESP3D/issues/249#issuecomment-475857956

Felix-ArduPilot, 2020-10-23:
Ok, that’s strange, because I flashed the module with the SD-card method. I plugged the WiFi module into the tft board and inserted an SD-Card with the correct files. I don’t think that would have worked with a wrong pinout, would it?
And when I use BTT FW, do I also have to change pinout?
Sorry for all the questions. I’m new to all the WiFi printing things.

luc-github, 2020-10-23:
when you flash MKS TFT it use MKS TFT pinout - it is normal 
The problem is whe you set MKS HLK-WIFI:1  MKS TFT use different  pins to communicate with module which are not the correct one for MKSTFT WIFI module

BTT FW for MKS TFT use correct pinout for MKS TFT WIFI module in TFT FW by default and so it is fine

Felix-ArduPilot, 2020-10-23:
Thanks for your great help!
I’m going to try BTT firmware then. Let’s see, if it works...

luc-github, 2020-10-23:
depending of mks tft you need a ported btt tft FW instead of original

Felix-ArduPilot, 2020-10-23:
Where would I get that for the MKS TFT28 V4.0?

luc-github, 2020-10-23:
check here https://github.com/darkspr1te/MKSTFT_Marlin_Touch and https://github.com/Uz45/BIGTREETECH-TouchScreenFirmware

https://github.com/darkspr1te/MKSTFT_Marlin_Touch/issues/6

luc-github, 2020-10-23:
Mine are TFT 32 so cannot confirm for TFT28

Felix-ArduPilot, 2020-10-23:
Thank you so much! It’s working now! :D I struggled a bit, because BIGTREE_TFT28_V3.0.26.x.bin isn’t working, but I also tried MKS_32_V1_4.26.x.bin as discussed in the last link you shared and that did the trick. I can now connect to the WiFi network and control the printer with the webinterface. Awesome!

I also tried uploading a printfile, but I got an “SD init failed”. But I’ve read somewhere, that you should only use sd-cards <8GB and my smallest card right now is 8GB. So I guess that might be the problem.

luc-github, 2020-10-23:
use properly formated card, use the tool from https://www.sdcard.org/downloads/formatter/

Felix-ArduPilot, 2020-10-23:
I used the tool, but that didn’t help. Strange thing is: I can see files that I placed on the SD-card. They show up on the printer screen. So the card is recognized. I just can’t upload files to it via WiFi.

luc-github, 2020-10-23:
you cannot upload to tft there is not such feature
only to printer sd card

Felix-ArduPilot, 2020-10-23:
What do you mean by printer sd card? The card that goes into the slot on the back of the tft 28 display or the USB-stick, that is connected to the display? I have installed both and I can see and access both through the printer menu. I just can’t upload files.

luc-github, 2020-10-23:
you cannot upload to tft (USB neither SD card) tft do not have upload function currently, only printer board with connected sd card reader
 if you printer board do not have sd card reader connected with sd card inside  then it cannot init the sd card

Felix-ArduPilot, 2020-10-23:
Ah okay. Thanks for clarifying that. I didn’t know that. So I’m going to install an sd card reader soon.
And except for the file upload everything is working now. Thank you sooo much for your help! I couldn’t have done that myself.

luc-github, 2020-10-23:
np I close issue then

closed 2020-10-23 by luc-github