luc-github/ESP3D #520: [BUG] 8.3 filename bug

victornpb, 2020-11-02:
**Describe the bug**
I'm uploading a file with a short name, and the WebUI is complaining about the name not being 8.3. I tried Uppercase/Lowercase. I'm not sure if its a FW bug or just the Web UI, I'm not seeing any errors in the devtools so I'm reporting it here.

I tried in 2 browsers, same results.

**To Reproduce**
Steps to reproduce the behavior:
1. Navigate to IP eg 192.168.0.26
2. Click on "Upload"
3. Choose a file ("BULBA.GCO")
4. See error "Please use 8.3 filename only."

**Expected behavior**
Upload the file

**Screenshots**
[image]
[image]

**ESP3D Firmware:**
UI: 2.1b67 / FW: 2.1.1.b8

- ESP core version: [2.5.2]
- Version:[e.g 2.1b4]
- Web UI Version:[e.g 2.1b14]  
- Wifi mode:[e.g. STA]  
- Flash method: [PlatformIO]
- Any change done in FW: NONE

**Target Firmware:**
 - Name: Marlin
 - Version 2.0
 
 tip: to get firmware info type `M115` in console    

**Board used (please complete the following information):**
 - MCU: ESP8266
 - Name: E01
 - Flash size: 4M

**Additional context**
Browser: Google Chrome Version 86.0.4240.111 (Official Build) (x86_64)
Browser: Safari Version 14.0 (15610.1.28.1.9, 15610)
OS: MacOS 10.15.7 (19H2) Catalina

luc-github, 2020-11-02:
previously it happen that the file extension was hidden on osx and the hidden extension was .gcode, which made the file name not 8.3 but BULBA.GCO.gcode
can you confirm by checking file properties?

victornpb, 2020-11-03:
Indeed, the file was previously named "foo bar.gcode" then I renamed to "foo.gco" but it was in fact renamed to "foo.gco.gcode" and it decide to hide the extension even tho it was set to show extension before 😕.

This issue could be either mitigated showing the file name in the error message like `The file named "%s" is not allowed!\nOnly 8.3 DOS is supported by marlin"` or something like that.

Another solution would be to prompt the user to type a short name instead of making then rename it.

Pseudo implementation:
```js
filename='Loren Ipsum Sit.gcode'

const is83 /^\w{1,8}\.[a-z0-9]{1,3}$/.test(filename)
if(!is83){
  filename = prompt('The original filename is too long, provide a 8.3 name', filename.substring(0,8)+'.g')
}

console.log('uploading...', filename)
```

luc-github, 2020-11-03:
yes but there is already this message only for this Marlin 'feature' , I think the best would be not having message at all if Marlin support long file names properly.
I have already explain the situation in : https://github.com/luc-github/ESP3D/issues/326  

I am happy you solved your name issue, and thank you for the suggestion, but I won't change anything in 2.1 webUI, 
SD serial file transfert is not yet done in 3.0 so I am hoping the long file name will be solved by then

Also to be honest, even the message would display the long filename : e.g.  `BULBA.GCO.gcode`,  I am confident people will still complain and say that they only typed  `BULBA.GCO' because of hidden extension
.
So instead of trying to workaround all possible users cases, and in this case Murphy law is very applicable,  I prefer not handle it at all and wait for long name support in Marlin - all others FW support long file names even Marlin forks https://github.com/luc-github/ESP3D/wiki/Firmware--support#sd-card-file-list

There are a lot of discussions about this short name feature in Marlin side and several PR ongoing but nothing is implemented yet.

luc-github, 2020-11-03:
I guess issue can be closed

closed 2020-11-03 by luc-github