luc-github/ESP3D #521: [BUG]Broken filesystem on esp8266 3.0

th0m4sek, 2020-11-08:
**Describe the bug**
File system is broken and webui can't be uploaded.

**To Reproduce**
Build new alpha for esp8266 and upload

**Expected behavior**

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

luc-github, 2020-11-08:
sorry I cannot duplicate - give proper info as bot said

th0m4sek, 2020-11-08:
I'm using ESP-12-F with 4M flash.
Firmware build by PlatformIO
ESP3D Firmware:

    ESP core version: [2.6.2]
    Version:[e.g 3.0.0.a69]
    Web UI Version:[none]
    Wifi mode:[e.g. AP]
    Flash method: [PlatformIO]
    Any change done in FW none

th0m4sek, 2020-11-08:
On the screen you can se also Status / does not exists! and traying to upload index.html.gz gives error write fail.

th0m4sek, 2020-11-08:
Ive tried recompiling with littlefs because i was getting plenty errors for spiffs. problem remains same but i notice that SDK used for building is not 2.6.2 `chip id: 49950 
CPU Freq: 160 Mhz 
free mem: 20.26 KB 
SDK: 2.2.2-dev(38a443e) 
flash size: 4.00 MB 
size for update: 1023.00 KB 
FS type: LittleFS 
FS usage: 1024.00 MB/1.00 GB 
baud: 115200 
sleep mode: none 
wifi: ON 
hostname: esp3d 
HTTP port: 80 
Telnet port: 23 
Ftp ports: 21, 20, 55600 
Websocket port: 8282 
ap: ON 
mac: 62:01:94:0B:C3:1E 
SSID: ESP3D 
visible: yes 
authentication: WPA2 
DHCP Server: ON 
ip: 192.168.0.1 
gw: 192.168.0.1 
msk: 255.255.255.0 
clients: 0 
sta: OFF 
mac: 60:01:94:0B:C3:1E 
i-time: OFF 
serial: ON 
notification: OFF 
Target Fw: unknown 
FW ver: 3.0.0.a69 
FW arch: ESP82XX 
`
Hope this will help some more

luc-github, 2020-11-08:
this esp8266 core 2.6.2 is not correct  - also on my side on platformIO  have compilation failed when ok under arduino IDE
also fs should be 2M instead of 1M - may be platformio.ini is outdated 
And using 1M I do not get same as you 
[image]

luc-github, 2020-11-08:
Ok looks like there is bad copy and paste in plateformIO - it use 4M flash no FS...
I will fix platformIO.ini

luc-github, 2020-11-08:
I have updated platformio.ini for flash size and esp core version to use git version
I have also commented timestamp by default until more tests on latest esp8266 core
So if SPIFFS it give a lot of warning because now deprecated but still ok 
[image]

using LITTLEFS give when using arduinoIDE  : 
[image]
But it seems using platformIO I still get SPIFFS and upload on LittleFS cannot be read- I will continue to investigate

luc-github, 2020-11-08:
Ok I cannot duplicate issue - anymore - I suspect a compiler cache pb when I changed SPIFFS with LITTLEFS
when cleaning previous files littlefs is fine in about and after upload ... keep watching anyway

th0m4sek, 2020-11-08:
I can confirm issue is solved.
Thanks

closed 2020-11-08 by th0m4sek

luc-github, 2020-11-08:
Thank you for the confirmation 😺