luc-github/ESP3D #522: [Question] a way to upload from Cura to esp3d?

kad, 2020-11-08:
I'm trying to understand, is there any way to use ESP3D somehow similarly to OctoPrint together with Cura?
I have no problems of compiling, flashing, configuring, connecting over, execute commands in WebUI... all of this is trivial.

But missing the final link: how to point cura that gcode file can be sent to ESP3D that will put it on some storage (onboard or TFT) and start printing that file. Really like octoprint experience: "Slice" button then -> "Upload to octoprint and print" button.

And unfortunately in all piles of videos/wiki/FAQ that I can google about esp3d it is all about "dummies guide on how to compile/install esp3d".

luc-github, 2020-11-09:
From general point of view:
currently I do not think there is a way to upload from esp3d to TFT SD  - at least I am not aware of it
for printer board  sd upload - which if you use serial communication is slow like hell - you can use something like `curl` to upload  in ESP3D 2.1 but authentication is not supported by this way 
something like : `curl -i -X POST -H "Content-Type: multipart/form-data" -F "data=@test.gco" http://mysuperserver/upload`

streaming gcode is also possible but I do not really recommend due to wifi latency and wifi communication delay that would bring some blobs.

A comment about same : https://github.com/luc-github/ESP3D/issues/275

About Cura:
There is no native support of cura communication protocol (I do not use cura, so I cannot comment on possibilities, sorry), you may add a pluggin on cura side to support TCP communication or to post the gcode to esp3d

There are some under going actions to add slicer in esp3d ui, but so far nothing available yet :  https://github.com/luc-github/ESP3D/issues/475

kad, 2020-11-09:
So, if transferring files from slicers to storage on the printer is not really a good option due to 115k serial line between esp3d and printer, sending command-by-command gcode from external entity to "orchestrate" printing is also not good idea, due to possible disconnects/latencies of wifi... that means that esp3d both serial-over-tcp and web ui can be used only for few things: 
- do some small movements
- convenient terminal to send commands to printer
- monitor current state (e.g. temperatures)

Anything else it can be used for?

For Cura: my question is not particularly in Cura per se, but about generic workflow:
1. load model into slicer
1. slice
1. from slicer transfer to entity that will orchestrate print
1. start execution of prepared gcode (print).

For step 3, classical mechanism is to save on card manually, then move that card to printer, and step 4 is "print from SD" on the printer.

With Cura/Prusa Slicer integration with OctoPrint, step 3 is just click a button of "send gcode to OctoPrint and start printing". Step 4 is executed by OctoPrint by communicating with printer over serial port.

So, rephrased question: is there are any combination of software that can use esp3d for similar steps 3 and 4, like it is currently implemented with OctoPrint?

luc-github, 2020-11-09:
all depend on how you use esp3d and on what 
if used on boards that allow direct SD access you can do file upload (Azteeg wifi / wifi back pack / ESP32 based printer board)

If only connected by serial currently it is for monitoring  / control printer - if associated with  wifi SD card,  upload speed is solved 

ESP3D works on 1$ chip, so of course you cannot expect it will be as  powerful  as octoprint that use PI board and linux, but for basic features it do the job IMHO 

I do not use octoprint so I will not be able to answer in detail for workflow - but ESP3D and ESP3D-WebUI is opensource so any contribution is welcome to make things possible ^_^

kad, 2020-11-09:
@luc-github thanks for your project, I understand limitations of the ESP platforms, just trying to understand on where potential limits of your project are.

So, assuming that if I use e.g. wemos d1 (esp8266) board with microSD shield (so, the storage is local to esp3d), or some esp32 board with has microSD card, will it be able to "orchestrate" print over serial connection to the printer from uploaded gcode?

If yes, then it will be different story on thinking how to adopt "octoprint-like" workflow with that setup.

luc-github, 2020-11-09:
to host GCODE on esp board sd and serve it to printer chich does not have access to it - the esp3d need  a GCODE host stream capability  which is planned but not ready yet : https://github.com/luc-github/ESP3D/issues/243

Also it should be hosted on esp32 dual core, not esp8266/esp32 single core, due to performance issue doing constant serial streaming  and handling the wifi in parallele

kad, 2020-11-09:
Ok, understood. Thank you @luc-github

closed 2020-11-09 by kad

luc-github, 2020-11-09:
@kad in addition there is this FYR https://github.com/luc-github/ESP3D/issues/523