luc-github/ESP3D #523: [FEATURE REQUEST]MKS WIFI support

luc-github, 2020-11-09:
MKS released the source code of wifi device : https://github.com/makerbase-mks/MKS-WIFI
After quick look the SD transfer code seems based on reprap one using SPI but the marlin part seems using specific pin WIFI_IO1_PIN  to strigger the pins usage (Marlin/src/lcd/extui/lib/mks_ui/wifi_module.cpp) so need to check deeper
the transfert is supposed to be 100KB which is far better than the slow M28/M29 protocol

So far current implementation is limited to MKS boards (Robin (s) and TFT)  so do not rush and expect a miracle in ESP3D for other boards yet

Currently compilation for ESP module  failed on my side  because I suspect it use DC42 esp core sources - not generic ones

I know LPC team ported DC42 sources to esp core 2.4.2  so it may be the case also for MKS - I did not checked

Even current sources are availables, there is a  current lack of documentation on usage - available chinese doc is supposed to be translated to english soon but it may just be a reference. Use google translate is working FYI.

This post is for raising, I do not have any MKS Robin board to test, conclusion: no ETA for esp3d implememtation yet - a lot of things to do already and no supported hardware - but if someone want to help to add it (not just test it) - contribution is welcome

luc-github, 2020-11-12:
after reading code, it seems it does not actually need custom code from reprap as it does not use SPI access
it use a pin to trigger a bypass in Marlin  to not consider data as GCODE command but as pure data  and so send package of 1KB which speed up transfert as it is pure file transfert not more ack needed for each command

I do not have any hardware to test / verify more  but it is pretty simple 

About MKS protocole to send IP and others informations,  I did not digged  a lot but seems well documented in excel file so should not be a problem to integrate it

luc-github, 2020-12-07:
Ok MKS Robin nano V2 received I can check in parallele with MKS TFT  + MKS WIFI

SimoneCarnio, 2020-12-14:
If I can help in some way, please tell me. I don't really know very well all this things, I'm writing a customized version of marlin+mks and just replaced mkswifi with esp3d and learning right now how this things work (I have a Nano Robin v1.1, v1.2 and v2.

luc-github, 2020-12-15:
@SimoneCarnio thank you - I wil start to work on it this week - I need first to understant the protocole and isolate the code from the fw code 
Then I will see how to add it in ESP3D - so not sure when it can be tested - the FW between MKS WIFI for TFT and the MKS WIFI for Robin board should not be too different but I need to read / dig in existing code first 
I will update when there is something to test

luc-github, 2020-12-16:
Reading at code MKS use actually 2 pins to control the transfer
GPIO 0 to send flag From ESP->Board
GPIO 4 to send flag From Board -> ESP

luc-github, 2020-12-29:
Ok I have now (I think) a good idea of MKS protocol thanks to doc and deep dig in code
* unlike the HLK-RM04 device which works in text mode the MKS ESP protocole use lot of binary flags and data
so every commands in/out to/ from MKS board need to be encoded/decoded and the trigger  is no more a return cariage `\n`but need to be head flag `0xa5`  `¥`and tail flag `0xfc` `ü` special chars so it is only for text files transfer and not using these chars
* the fast upload use the encoded protocol and also 2 pins one as input to know if printer is ready to receive, one as output to notify printer ESP want to send data - this is part as communication protocol
* the MksWIFI plugin use telnet port 8080 - the communication itself seems plain text as it works wirh current ESP3D FW but SD listing and of course upload - That should be supported in porting
* there is an UDP server using port 8989 and no idea of it usage - so I won't support it
* there is a cloud service using port 12345, but all code has been removed from public sources - so I won't support it
* the wifi infos are send to printer every 10s
* the printer/tft can query the list of hotspots from ESP3D
* some specific GCODE may need to be used for selecting USB Disc / SD and know the status but they are documented in doc

So the porting will be : 
* to write the coding decoding service and integrate it in ESP3D - as it will affect the command interpreter and the way it work - it will be a configuration flag and not a setting - need to be defined at compilation time
* Add fast upload support that use the above Msk protocol
* sort / ignore messages coming from MKS board/TFT for unsupported services 
* add SD file listing format for the MKS WiFI pluggin
* add Temperature format for MKS Wifi pluggin as it seems the MKS FW rewrite the M105 output - TBC 
on my repetier printer the temperatures are not detected, but it may due to repetier M105 format, others commands but sd related seems working 
[image]

Note: I have implemented the wifi info using the MKS protocol and send every 10s - the MKS Robin is happy and show the correct IP status and says it is connected - when ESP is restarting the status go to disconnected - then connected when esp is back online so detection is ok which is good start ^_^

luc-github, 2020-12-29:
Ok  starting to get some communication 
[image]

makerbase-mks, 2020-12-30:
@luc-github ,Hello, UDP server using port 8989 is for wifi module discovery in LAN, so the PC or phone can scan the wifi modules in LAN, no need to input the IP manually.

UDP port: 8989
Wifi module polling monitor, if a string is received:
```
mkswifi
```
The wifi modulereturns:
```
mkswifi:$printer_unique_ID, $internal_network_IP\n
```
The $printer_unique_ID is a string of 20 bytes, such as:
```
mkswifi:HJNLM0006001941A4855,192.168.1.123\n
```

luc-github, 2020-12-30:
@makerbase-mks thank you for the clarification - so it is kind of discovery protocol for MKS  App 
Because ESP3D has already mDNS and SSDP it may not necessary in ESP3D and it, also not to include it in ESP3D may avoid to confuse MKS App

luc-github, 2020-12-30:
According https://youtu.be/0GKT0hw_K-4?t=296 the cloud service  generate a qrcode in UI - because Cloud is not available in public FW I wanted to use it to help user to connect to his printer by scanning it and get the IP/port but seems  lv_lib_qrcode is not part of Marlin code so I guess also part of private MKS code - so cannot display any QRCODE in Marlin version

I do not have bandwidth to work on Marlin for this  - if anyone want to do it - just poke me and I will adjust ESP3D data accordingly - today I keep original cloud values Host/Port and module ID is set to 12345 as not used

Another nice to have in Marlin, would be to display the WIFI FW version, but same a above - so feel free to implement it in Marlin extUI - in ESP3D the version is `ESP3D_3.XXXXXX` according esp3d version so code is ready

Any suggestion let me know

SimoneCarnio, 2020-12-30:
All "mks" stuffs is practically in the bugfix branch, on my version of the marlin+mks firmare I removed the cloud features. don't know if that lib can be used in classic marlin, but it can on the mks (lvgl) version. From the mks interface you can scan and connect to Wi-Fi, maybe this could be more interesting than showing the qrcode. So If I understand how esp3d work, can code something on my firmware and then make a pull request on the marlin repo.

luc-github, 2020-12-30:
@SimoneCarnio  I have checked Marlin github 2.0.7.2 and bugfix and cannot see where is the qrcode generation part - also cloud_hostUrl is not used actually - so I supposed the code is not there - if you can point me where is the code that generate the qrcode I can check

SimoneCarnio, 2020-12-30:
There is the qrcode generation, not the code for cloud, but there are the "pages" to see the qrcode. It is in the bugfix branch not stable. I Am out of Office now and driving, cannot check right now ￼

filimonic, 2020-12-30:
Take a look at this repo: https://github.com/Sergey1560/Marlin_FB4S
This guy ported MKS WIFI to marlin before MKS opened firmwares

luc-github, 2020-12-30:
@filimonic  but this part is Marlin part  it looks enhanced version of Marlin but not uising LVGL UI part but doing same and repo has also marlin part https://github.com/Sergey1560/Marlin_FB4S/tree/FB4S_WIFI/Marlin/src/lcd/extui/lib/mks_ui so I am not sure what to look at

@SimoneCarnio take your time - it is not hurry - let me know when you have time where is the qrcode code part - it does not pop up to my eyes where ever it is 😵

SimoneCarnio, 2020-12-30:
@filimonic that one is a version of marlin optimized for the Flying Bear Ghost and with the mks ui merge (probably always from the bugfix branch)

@luc-github  I'm sorry, too many repos checked this week, the "cloud" part is only on the mks repo.  (https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/blob/master/Marlin/src/lcd/extui/lib/mks_ui/draw_cloud_bind.cpp).   Anyway, if I understand how to do it (how to communicate with esp3d), I can make an esp3d integration and when esp3d is connected to Marlin show a qrcode with a "wifi qrcode"  like this one and help the user join esp3d in it is in AP mode

[image]

luc-github, 2020-12-30:
@SimoneCarnio yes too many repositories - it weird this is not merged in Marlin repository
you do not need to communicate with ESP3D you just need to replace in 
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/blob/master/Marlin/src/lcd/extui/lib/mks_ui/draw_cloud_bind.cpp#L224-L225
`cloud_para.id`
with an http address using
```
  cloud_para.hostUrl
  cloud_para. port
```

and it should work

SimoneCarnio, 2020-12-30:
Ok, sorry if I understand something, I didn't check that things yet, I'm still working on other functionalities

luc-github, 2020-12-30:
I have already done the modifications in ESP3D to put  ESP3D IP in cloud_host and web server port in cloud port

I will push the changes tonight - I still have one message to handle to push the code

luc-github, 2020-12-30:
Protocol  is implemented but upload
next step : fast upload implementation - I will test with curl once implemented
next step 2 : WebUI update to support the SD upload and listing 
next  step 3 : test on MKS TFT to check USB support

luc-github, 2021-01-04:
Got delay due to a bug in latest esp8266 core about upload ( which is now fixed), and I have to dig more in Marlin code and mks wifi original code to check the weird usage of content-lenghT as File lenght, which even can fit for text fileS, is totally innacurate , and finally found  that the file lenght info is actually ... not used orz...

So now everything is clarified now (I hope) - I can finalize the transfert part 

Also I will use this Marlin version for reference for my Robin Nano 2: https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware as it seems the most up to date  -  I will recompile it to verify and test code with it

luc-github, 2021-01-07:
Fast upload is implemented ~110KB/S using curl

[image]

```
C:\Users\luc\Documents\GitHub\ESP3D-WEBUI>curl -v -F "file=@CUBE01.GCO" http://192.168.1.108/upload
*   Trying 192.168.1.108...
* TCP_NODELAY set
* Connected to 192.168.1.108 (192.168.1.108) port 80 (#0)> POST /upload HTTP/1.1
> Host: 192.168.1.108
> User-Agent: curl/7.55.1
> Accept: */*
> Content-Length: 2090034
> Expect: 100-continue
> Content-Type: multipart/form-data; boundary=------------------------d0a76c45135fe417
>
* Done waiting for 100-continue
< HTTP/1.1 200 OK
< Content-Type: text/plain
< Content-Length: 15
< Connection: close
<
{"status":"ok"}* Closing connection 0

C:\Users\luc\Documents\GitHub\ESP3D-WEBUI>
```

I may need to cleanup the code but it is working 

Next step is upload using web-UI

luc-github, 2021-01-07:
So ESP3D 3.0
Firmware: 3.0.0.a80  
WebUI: 3.0.0.77 
is now working with my MKS Robin Nano V2 and the MKS WIFI 2.0 😸 

Next Step is MKS TFT + MKS WIFI 1.0 
I need to reflash my MKS TFT32_L V2.0 with MKS FW as I used the BTT FW version on it
On this one need to verify :
* the pin setup 
* the transfer on SD 
* the transfer on USB (TBC)

luc-github, 2021-01-08:
Communication is ok - the TFT is not connected to any board for the moment

[image]

WiFi status is ok too   

[image]

Upload is working on SD  - the speed difference with MKS Robin Nano is due to SD card quality - on TFT it is a brand new High speed SD
[image]

I did not checked USB yet 

So far so good for the moment

About discussion on QR Code:
Using MKS TFT FW + MKS WIFI 1.0 FW I got qr code:
[image]

But none of my QR code app can read it - may be due to blue dots ? 

I have also uncommented `display_qrcode((uint8_t *)cloud_para.id);` in mks marlin https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/blob/master/Marlin/src/lcd/extui/lib/mks_ui/draw_cloud_bind.cpp#L164

but just got this :
[image]

I won't investigate on this part I think - so anyone - feel free

luc-github, 2021-01-08:
Ok seems module id is not correct if correct the qrcode is readable on TFT :
[image]

For Marlin it is not the issue as I used a 21 chars fake  module id

makerbase-mks, 2021-01-08:
@luc-github .Actually, you can generate the ID : HJNLM000+MAC string of module

luc-github, 2021-01-08:
@makerbase-mks ok I will add this then thank you

luc-github, 2021-01-08:
With right ModuleID even Marlin QR Code generation is working 
[image]

luc-github, 2021-01-08:
So what left is USB content listing:
`M998 0` is the command to select the USB and `M998 1` is the command to select SD
`M20` is the one to list what ever the target
But seems not working, only SD is listed even usb is seen when checking TFT manually
`M998 0` and `M99 1` both also switch TFT UI  File system to SD when checking on it

[image]

Not sure where the issue come from as sending commands from WebUI or telnet is same  behavior - only SD

luc-github, 2021-01-08:
Ok looking better at MSK FW code (TFT and  MKS Wifi) :
actually commands M998 is half a setting and half a command to list FS content

`M998 0` do `M998 0` and `M20 0:` will set USB as FS and list USB content
`M988 1` do `M998 1` and `M20 1:` will set SD as FS and list SD content

`M20` only is SD list what ever M998 0 was set or not : will set SD as FS and list SD content

Doing `M998 0` then `M20 0:` allow to upload on USB 

So I will implement the mecanism in web UI now everything is clear 😸

luc-github, 2021-01-08:
also need to add support of MKS specific commands output
```
>M27
M27 0

> M994
M994 /CUBE.GCO;210212

>M992
M992 00:01:58

> M997
M997 PRINTING
```

Note:
I found a bug in Marlin  - `M27` only show `M27 0` even the status on tft is not 0% like 3%, M27 on Serial is correct 
here https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/blob/master/Marlin/src/lcd/extui/lib/mks_ui/draw_printing.cpp#L442
need to uncomment and replace `gCurFileState` by `uiCfg`

`uiCfg.totalSend = rate;`

then M20 give right output on WiFi 

Also the QRCode commented -https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/blob/master/Marlin/src/lcd/extui/lib/mks_ui/draw_cloud_bind.cpp#L164
which uncomment display the correct qrcode

ETE-Design, 2021-01-08:
@luc-github Have an MKS Robin Lite v1.1 there the Wifi Module is direct on the Mainboard... I intended to use your firmware ESP3D 3.0 On the Wifi Module to have all features, the Mainboard have both WIFI_IO1_PIN + WIFI_IO2_PIN... So can you somehow do so your firmware also support Fast Upload Support... Or is it already implemented?

luc-github, 2021-01-08:
should work fw is same across module

ETE-Design, 2021-01-08:
@luc-github So now your firmware support for all boards as long the pin is presented?

luc-github, 2021-01-08:
yes MKS Wifi modules are all using same pins

ETE-Design, 2021-01-08:
@luc-github Just wasen't sure on I had to use MKS Firmware to use that feature, but it sounds like your firmware could also be used :-)

luc-github, 2021-01-09:
Just found a glich :
the MKS Robin only list root and hide sub directories contents - and there is no way to access them - but using Serial M20 it list all SD content - so I think it is a bug in Marlin here: https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/blob/master/Marlin/src/lcd/extui/lib/mks_ui/wifi_module.cpp#L702-L742
The path variable is not used  and subdir is just displayed as .DIR
the TFT allows to list all subdirectories - and so access / upload any location

I need to adjust FW as currently the MKS upload was limited to root because I thought it was a limitation - but it seems more a bug on robin as tft is ok

Which bring another question : how to specify path when using CURL - as it only use filename not path 
need to add a rpath variable detection  to not colide with path
this also should allow to select target SD or USB as by default it is active FS but curl has not way to set it

something like:
`curl -v -F "file=@CUBE01.GCO" -F "rpath=USB:mysub" http://192.168.1.108/upload`
 but I need to test it

luc-github, 2021-01-11:
Note: to specify path need to put path definition before filename as post file is processed as soon as received, so rpath variable must be received before file

`curl -v -F "rpath=mysub" -F "file=@CUBE01.GCO" http://192.168.1.108/upload`
or use get variable
`curl -v -F "file=@CUBE01.GCO" http://192.168.1.108/upload?rpath=mysub`

I also need to remove device target and handle a pre `M998 0` /`M998 1`  command first to select the right target, this is only for curl type command as webUI will do it by itself

But not TFT does not support `M30 filename` to delete file neither `M32 dir name` to create dir name

luc-github, 2021-01-11:
WebUI support for MKS upload /print to/from USB/SD , also display print status
*  SD of Robin Nano V2
[image]
* MKS TFT SD subdir
[image]
Note:
-  Dir creation is a not supported by Marlin neither TFT
- File deletion is not supported by TFT

luc-github, 2021-01-11:
I think I am done for this part

closed 2021-01-11 by luc-github