luc-github/ESP3D #525: [FEATURE REQUEST]allow to add picture in notification

luc-github, 2020-11-14:
ability to attach picture or document to notification
email
telegram
line
pushover

one usage would be to enclose a camera capture per user feedback TBC

it is definitly not a priority task so no time frame defined

liqijian101, 2023-04-11:
hi luc,do you have time to add this feature now? I tried to add it, but I failed. This is a very useful feature.

liqijian101, 2023-04-11:
//Send message
    camera_fb_t * fb = NULL;
    size_t _jpg_buf_len = 0;
    uint8_t * _jpg_buf = NULL;
    fb = esp_camera_fb_get();
    _jpg_buf_len = fb->len;
    _jpg_buf = fb->buf;
    Notificationclient.printf("From:ESP3D<%s>\r\n",_settings.c_str());
    Notificationclient.printf("To: <%s>\r\n",_settings.c_str());
    Notificationclient.printf("Subject: %s\r\n\r\n",title);
    //Notificationclient.println(message);
    Notificationclient.println("Content-Type:image/jpg;name=ESP3D.jpg\r\n");
    Notificationclient.println("Conten-Transfer-Encoding:base64\r\n\r\n");
    Notificationclient.print(base64::encode((const char *)_jpg_buf));
    esp_camera_fb_return(fb);
    fb = NULL;
    _jpg_buf = NULL;
    log_esp3d("Send final dot");
    //Send Final dot
    Notificationclient.print(".\r\n");

luc-github, 2023-04-11:
No sorry, it is still in (huge) todo list

liqijian101, 2023-04-13:
HI Luc, I have modified the notifications_service.cpp file and now I can send images via email. As long as I send the notification, the images will be sent. Can you make this feature more perfect?

/*
  notifications_service.cpp -  notifications service functions class

  Copyright (c) 2014 Luc Lebosse. All rights reserved.

  This code is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This code is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with This code; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
*/
//Inspired by following sources
//* Line :
// - https://github.com/TridentTD/TridentTD_LineNotify
// - https://notify-bot.line.me/doc/en/
//* Pushover:
// - https://github.com/ArduinoHannover/Pushover
// - https://pushover.net/api
//* Email:
// - https://github.com/CosmicBoris/ESP8266SMTP
// - https://www.electronicshub.org/send-an-email-using-esp8266/
//* Telegram
// - https://medium.com/@xabaras/sending-a-message-to-a-telegram-channel-the-easy-way-eb0a0b32968

#include "../../include/esp3d_config.h"
#ifdef NOTIFICATION_FEATURE
#include "notifications_service.h"
#include "../../core/settings_esp3d.h"
#include "../../core/esp3doutput.h"
#include "../network/netconfig.h"
#include "../camera/camera.h"
#include <esp_camera.h>
#include <WiFiClientSecure.h>

#if defined( ARDUINO_ARCH_ESP8266)
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <libb64/cdecode.h>
#endif //ARDUINO_ARCH_ESP8266

#if defined(ARDUINO_ARCH_ESP32)
#include <WiFi.h>
#include <HTTPClient.h>
extern "C" {
#include "libb64/cdecode.h"
}
#endif //ARDUINO_ARCH_ESP32
#if defined (HTTP_FEATURE) || defined(WS_DATA_FEATURE)
#include "../websocket/websocket_server.h"
#endif //HTTP_FEATURE || WS_DATA_FEATURE
#if defined (DISPLAY_DEVICE)
#include "../display/display.h"
#endif //DISPLAY_DEVICE
#include <base64.h>

#define PUSHOVERTIMEOUT 5000
#define PUSHOVERSERVER "api.pushover.net"
#define PUSHOVERPORT    443

#define LINETIMEOUT 5000
[... 554 lines cut ...]
    }
    _started = res;
    return _started;
}
void NotificationsService::end()
{
    if(!_started) {
        return;
    }
    _started = false;
    _notificationType = 0;
    _token1 = "";
    _token1 = "";
    _settings = "";
    _serveraddress = "";
    _port = 0;
}

void NotificationsService::handle()
{
    if (_started) {
    }
}

#endif //NOTIFICATION_FEATURE

luc-github, 2024-10-21:
I close issue due to out of scope of ESP3D

closed 2024-10-21 by luc-github