luc-github/ESP3D #526: [BUG] Failed to communicate with FW!

MillerRen, 2020-11-16:
**Describe the bug**
WebUI-3.0 show me "Failed to communicate with FW!". The request "http://192.168.0.1/command?cmd=%5BESP800%5Dtime=2020-11-16-19-55-38" respond empty or timeout. ESP3D-2.1 is ok.

**To Reproduce**
Steps to reproduce the behavior:
1. download WEBUI-3.0 index.html.gz to ESP3D /data directory
2. upload firmware esp3d-3.0 to nodemcu dev board.
3. connect wifi "esp3d"
4. redirect not automicly. input 192.168.0.1 manualy.
5.web ui show me that error. serial console monitor can not response also.

**Expected behavior**
webui-3.0 work well as webui-2.1

**ESP3D Firmware:**
- ESP core version: 2.5.2
- Version:3.0
- Web UI Version: 3.0
- Wifi mode: not sure 
- Flash method: platformio
- config platformio platform=espressif8266. 

**Target Firmware:**
 - Name: [e.g. Marlin]
 - Version [e.g. 2.0]
 

**Board used (please complete the following information):**
 - MCU: esp8266
 - Name: nodemcu
 - Flash size: 4M

luc-github, 2020-11-16:
check what [ESP420]plain give as output
also what is the version on top left of 
`http://<your ip>?forcefallback=yes`
looks like you are not using esp3d 3 0 FW but 2.1 FW with webUI 3.0

MillerRen, 2020-11-16:
still not work. any wrong？ here is the platformio messages:

> Executing task in folder ESP3D-3.0: C:\Users\renyufei\.platformio\penv\Scripts\pio.exe run --target upload <

Processing esp8266dev (platform: espressif8266; board: esp12e; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (2.6.2) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 160MHz, 80KB RAM, 4MB Flash       
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - tool-mklittlefs 1.203.200522 (2.3)
 - tool-mkspiffs 1.200.0 (2.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Converting esp3d.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <EEPROM> 1.0
|-- <Ticker> 1.0
|-- <Wire> 1.0
|-- <lv_arduino> 2.0.3
|-- <ESP8266_SSD1306> 4.0.0
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <TFT_eSPI> 1.4.11
|   |-- <SPI> 1.0
|-- <ESP8266SdFat> 1.1.0
|   |-- <SPI> 1.0
|-- <LittleFS(esp8266)> 0.1.0
|-- <SD(esp8266)> 2.0.0
|   |-- <SDFS> 0.1.0
|   |   |-- <ESP8266SdFat> 1.1.0
|   |   |   |-- <SPI> 1.0
|   |   |-- <SPI> 1.0
|-- <ESP32SSDP> 1.1.1
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266SSDP> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266-Arduino-Lua> 0.0.30
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <BMx280MI> 1.2.0
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <DHT sensor library for ESPx> 1.0.6
|-- <WebSockets> 2.3.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <Ethernet(esp8266)> 1.0.4
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <Hash> 1.0
Building in release mode
Compiling .pioenvs\esp8266dev\src\esp3d.ino.cpp.o
Retrieving maximum program size .pioenvs\esp8266dev\firmware.elf
Checking size .pioenvs\esp8266dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [======    ]  57.2% (used 46852 bytes from 81920 bytes)
Flash: [======    ]  55.2% (used 576268 bytes from 1044464 bytes)
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM7
Uploading .pioenvs\esp8266dev\firmware.bin
esptool.py v2.8
Serial port COM7
Connecting........_____....._____.
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 8c:aa:b5:c6:3f:6d
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 580416 bytes to 414774...

Writing at 0x00000000... (3 %)
Writing at 0x00004000... (7 %)
Writing at 0x00008000... (11 %)
Writing at 0x0000c000... (15 %)
Writing at 0x00010000... (19 %)
Writing at 0x00014000... (23 %)
Writing at 0x00018000... (26 %)
Writing at 0x0001c000... (30 %)
Writing at 0x00020000... (34 %)
Writing at 0x00024000... (38 %)
Writing at 0x00028000... (42 %)
Writing at 0x0002c000... (46 %)
Writing at 0x00030000... (50 %)
Writing at 0x00034000... (53 %)
Writing at 0x00038000... (57 %)
Writing at 0x0003c000... (61 %)
Writing at 0x00040000... (65 %)
Writing at 0x00044000... (69 %)
Writing at 0x00048000... (73 %)
Writing at 0x0004c000... (76 %)
Writing at 0x00050000... (80 %)
Writing at 0x00054000... (84 %)
Writing at 0x00058000... (88 %)
Writing at 0x0005c000... (92 %)
Writing at 0x00060000... (96 %)
Writing at 0x00064000... (100 %)
Wrote 580416 bytes (414774 compressed) at 0x00000000 in 36.8 seconds (effective 126.0 kbit/s)...
Hash of data verified.

Leaving...
Soft resetting...
================================================ [SUCCESS] Took 49.12 seconds ================================================

Environment    Status    Duration
-------------  --------  ------------
esp8266dev     SUCCESS   00:00:49.116
================================================= 1 succeeded in 00:00:49.116 =================================================

[image]

MillerRen, 2020-11-16:
may be it mixed 2.0 and 3.0 webui in spiffs partation. I will swipe it by other tools. Thanks a lot.
I am a front end web developer. I want to known how it works. It maybe make a little help to improve the WebUI.

MillerRen, 2020-11-16:
got this.
[image]

MillerRen, 2020-11-16:
any ajax request is 404

luc-github, 2020-11-16:
sorry I  was out
ok - as you use nodemcu please check using a serial monitor the output of `{ESP420]plain` command
I suspect signal issue 

about WebUI - I am not web designer, the 3.0 will definitly hurt your eyes if you are one  ^_^
It is based on Preact and bottstrap 4.0  -  but as I am not preact dev, and I learned by myself these 8 month, the result is - webUI is working but style is not the preact one...

There is already a preact rewrite ongoing for 3.1 to be a real preact code: d it will use spectre instead of bootstrap - that said looks should be quite similar but better - the test / discussion branch is here https://github.com/luc-github/ESP3D-WEBUI/tree/3.0-spectre-poc all is done by @alxblog 

@alxblog  has lot of patience teaching me the `Preact mind` to code accordinly 
feel free to join discord channel to chat with me and alex  if you want to participate

luc-github, 2020-11-17:
@MillerRen did you sorted your issue ?

MillerRen, 2020-11-17:
@luc-github Not yet. I need learn more about esp chipset and libraries.

luc-github, 2020-11-17:
it shoud work out of the box - unless you did some code modification

MillerRen, 2020-11-18:
hi @luc-github It works. I erase flash by platformio in vscode. It seems like that some  lagacy files of esp3d-webui 2.0 effect.

closed 2020-11-18 by MillerRen