luc-github/ESP3D #528: [FEATURE REQUEST]TFT support

luc-github, 2020-11-17:
It was in my todo list and even did a prototype but now it exists so I will use it as reference :
https://www.rotrics.com/collections/add-ons-accessories/products/3-5-inch-touchscreen
Current TFT support is too basic in ESP3D and need a full rewrite

@MichaelDu9226 did a great first porting https://github.com/MichaelDu9226/ESP3D

But due to my poor current implementation of UI manangement in ESP3D - performances are not enough to give smooth result

so defintly need a rewrite :
* GCODE host oriented 
* TFT UI oriented|
* Touch oriented 

I need to study how BTT TFT / MKS TFT / Marlin MKS extUI are implemented and may be do some kind of porting 

Using LVGL for TFT is definitly the base

One issue I see is the images in UI these can use  a lot of flash space so ESP32 with 8M flash is a must (4MB for image storage for TFT UI)

Current UI of rotrics is B&W but hope can introduce color UI even with the limited space

Why not use the SD for storage ? I think it is not safe but one alternative would be to load fall back UI in flash 
so if any custom icons are in sd  it could take over - but need to see the performance impact

From another point of view there are 8M PSRAM but on arduino only 4M are available, moving to IDF is not really an options as too many arduino lib would need to be ported to idf - so let's stick to arduino

This will be a big project once ESP3D 3.0 is done (including all dependent projects) as it will be a base for it also

Any suggestion is welcome

ETE-Design, 2021-01-08:
@luc-github When you start working on it I can help you test it on BigTreeTech TFT35 E3, a really common used TFT on a lot of printers... But they have their own MCU so need seperate implementation I think, but you that way also save space for the GUI part on the ESP... Also have GPIO0 and GPIO2 aviable...

luc-github, 2021-01-08:
you misundeetood this topic is esp32 as main mcu for tft
not using stm32 based tft

ETE-Design, 2021-01-08:
@luc-github Seems like I did, just read the title "TFT Support", and then read BBT TFT, didn't know they make any ESP32 based TFT...

luc-github, 2023-01-01:
closing as now there is https://github.com/luc-github/ESP3D-TFT

closed 2023-01-01 by luc-github