luc-github/ESP3D #530: Marlin 2.0 Binary File Transfer?  [2]

Temtaime, 2020-11-21:
Hello.
Seems new protocol is pretty useful.
Is there any plans for implementing it?
Are PRs welcome ?
Thanks for your work.

luc-github, 2020-11-21:
same as https://github.com/luc-github/ESP3D/issues/327

and so far I did not saw any change in MArlin side - so it is still experimental

Temtaime, 2020-11-21:
Are PRs for experimental features welcome?
Can i implement it and have a chance get it merged?

luc-github, 2020-11-21:
you can do PR but I won't garanty it will be merged if no test is done, 
also as mentionned https://github.com/MarlinFirmware/Marlin/pull/14817 current protocol is not stable and subject to change which means better to keep PR open until it is stable 

also this means it must be limited to Marlin 2.0 in code - so need to add a way to differenciate Marlin 2.X from 1.X  and other Marlin flavors (custom / Marlin kimbra/ etc) which is not yet supported 

So finally 2.1 is not right target as code is almost frozen so it you want to do PR please use 3.0 branch which does not support yet serial transfert

there is another binary transfer posibility which is considered stable but I did not check if it can be used yet 
https://github.com/luc-github/ESP3D/issues/523

luc-github, 2020-11-24:
I close issue if no update and as duplicate as https://github.com/luc-github/ESP3D/issues/327

closed 2020-11-24 by luc-github