luc-github/ESP3D #531: [Hardware]ESP32 with ESP01 form factor

luc-github, 2020-11-30:
There are several devices that use ESP01 to add ESP3D, as ESP3D is growing with new features the ESP01S with 1M flash become too small to support ESP3D web update and new features.
The solution to upgrade the flash chip is validated but not really easy for most of people with lmited soldering skill (like me 😳 )

If you compare ESP01 dimensions 14.3mm X 24.8mm
[image]

with and ESP32 with 16MB flash dimensions 18mm x 25.5mm,   
we can imagine a final pcb around 20mm x 30mm, which still should fit existing space used by ESP01 IMHO
[image]

All pins could be compatible - not sure how to handle RST one actually, ad bridge GPIO 15 to GND would avoid boot garbage
[image]

About cost :
* ESP01S 1M: 1$~2$
* ESP32 16M: 2$~3$ + pcb cost (1$ ?) = 3$~4$ (?)    

Difference is not big but benefits are : more memory / more flash / more cores (x2)

I feel soldering ESP32 is easier than rework the memory for limited soldering skill people if come DIY 

One thing I am not able to evaluate is the power consumption of ESP8266 vs ESP32, but I guess there are some impact - so need to be tested.

I am not electronician, so please feel free to comment, I did not see such product yet - may be there is a reason ? 

Adfditionnaly if there is a way to allow access with some header the unused pins while keeping small form factor,  it would be a great upgrade.

alxblog, 2020-11-30:
About unused pins, 
HMUO, FFC connector with flat cable and breakout board might be a good way to deal with it.

vivian-ng, 2020-12-02:
Hi, maybe check out the [TinyPico](https://unexpectedmaker.com/tinypico), which is a tiny ESP32 development board. Anyway, it is definitely possible to just slap a ESP32 module to a PCB breaking out the GPIO0, RX, TX, 3.3V and GND pins if compatibility with the connector for ESP-01 module is critical. The thing is that you will still need to find a way to program the ESP32 (like the ESP-01, using a USB-UART device).

luc-github, 2020-12-02:
@vivian-ng yes, the critical part  is the connector and size to fit existing used socket, to be a pin to pin replacement, but the ESP32 seems not having a reset pin unlike ESP8266 - I do not see it

closed 2020-12-11 by luc-github