luc-github/ESP3D #532: [BUG]<does not upload files>

lidas1234, 2020-12-01:
Hello, I have a problem I loaded everything by following the guide and installing the most updated software, I connect to the printer I see the sd with the files inside it but if I go to load a file from the computer and press the load command nothing happens, nothing happens. it should open the window asking me which is the file to upload and then upload it, what can I do thanks

**ESP3D Firmware:2.1.1
- ESP core version: [2.5.2]
- Version:[e.g 2.1b4]
- Web UI Version:[e.g 2.1b57]  
- Wifi mode:[e.g. STA]  
- Flash method: [Arduino Ide / PlatformIO]
- Any change done in FW
 
 tips: to get ESP3D informations type `[ESP420]plain`    
       to get WebUI version check your menu header    

**Target Firmware:**
 - Name: [Marlin]
 - Version [2.0]

luc-github, 2020-12-01:
please give the output of [ESP420]plain

lidas1234, 2020-12-01:
Sorry I didn't understand, what should I do and where to find it, thanks

luc-github, 2020-12-01:
yes you did not filled the template with any options to allow me to help you 
2 ways :
1 - go to webUI terminal and type `[ESP420]plain` in command and give me the result also give me the M115 result
[image]

2 - screen the esp3d info page
[image]

lidas1234, 2020-12-01:
[ESP420]
{"chip_id":"5188827","cpu":"80","freemem":"37.30 KB","SDK":"2.2.2-dev(38a443e)","flash_size":"1.00 MB","update_size":"369.89 KB","spiffs_size":"228.20 KB","baud_rate":"250000","sleep_mode":"Modem","channel":"11","phy_mode":"11g","web_port":"80","data_port":"8888","active_mode":"AP (4A:3F:DA:4F:2C:DB)","ap_ssid":"ESP3D","ssid_visible":"Yes","ssid_authentication":"WPA","ssid_max_connections":"4","ssid_dhcp":"Started","ip":"192.168.0.1","gw":"192.168.0.1","msk":"255.255.255.0","connected_clients":[{"bssid":"00:08:CA:33:C7:B4","ip":"192.168.0.100"}],"disabled_mode":"STA (48:3F:DA:4F:2C:DB) is disabled","captive_portal":"Enabled","ssdp":"Enabled","netbios":"Disabled","mdns":"Enabled","web_update":"Enabled","pin recovery":"Disabled","autentication":"Disabled","target_fw":"Marlin","M117_output":"Enabled","Serial_output":"Enabled","Websocket_output":"Enabled","TCP_output":"Enabled","fw":"2.1.1.b8 ESP8266/8586"}
{"chip_id":"5188827","cpu":"80","freemem":"37.12 KB","SDK":"2.2.2-dev(38a443e)","flash_size":"1.00 MB","update_size":"369.89 KB","spiffs_size":"228.20 KB","baud_rate":"250000","sleep_mode":"Modem","channel":"11","phy_mode":"11g","web_port":"80","data_port":"8888","active_mode":"AP (4A:3F:DA:4F:2C:DB)","ap_ssid":"ESP3D","ssid_visible":"Yes","ssid_authentication":"WPA","ssid_max_connections":"4","ssid_dhcp":"Started","ip":"192.168.0.1","gw":"192.168.0.1","msk":"255.255.255.0","connected_clients":[{"bssid":"00:08:CA:33:C7:B4","ip":"192.168.0.100"}],"disabled_mode":"STA (48:3F:DA:4F:2C:DB) is disabled","captive_portal":"Enabled","ssdp":"Enabled","netbios":"Disabled","mdns":"Enabled","web_update":"Enabled","pin recovery":"Disabled","autentication":"Disabled","target_fw":"Marlin","M117_output":"Enabled","Serial_output":"Enabled","Websocket_output":"Enabled","TCP_output":"Enabled","fw":"2.1.1.b8 ESP8266/8586”}
M115
FIRMWARE_NAME:Marlin bugfix-2.0.x (Nov 17 2020 22:13:49) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Anycubic I3M EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:RUNOUT:0
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:1
Cap:CASE_LIGHT_BRIGHTNESS:1
Cap:EMERGENCY_PARSER:1
Cap:PROMPT_SUPPORT:0
Cap:SDCARD:1
Cap:AUTOREPORT_SD_STATUS:1
Cap:LONG_FILENAME:1
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:0
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0
area:{full:{min:{x:-5.0000,y:0.0000,z:0.0000},max:{x:220.0000,y:215.0000,z:200.0000}},work:{min:{x:-5.0000,y:0.0000,z:0.0000},max:{x:220.0000,y:215.0000,z:200.0000}}}
ok

> Il giorno 01 dic 2020, alle ore 09:45, Luc <notifications@github.com> ha scritto:
> 
> yes you did not filled the template with any options to allow me to help you
> 2 ways :
> 1 - go to webUI terminal and type [ESP420]plain in command and give me the result also give me the M115 result

luc-github, 2020-12-01:
please type 
[ESP420]plain 
for readibility

lidas1234, 2020-12-01:
[ESP420]
{"chip_id":"5188827",
"cpu":"80",
"freemem":"37.06 KB",
"SDK":"2.2.2-dev(38a443e)",
"flash_size":"1.00 MB",
"update_size":"369.89 KB",
"spiffs_size":"228.20 KB","baud_rate":"250000",
"sleep_mode":"Modem",
"channel":"11",
"phy_mode":"11g",
"web_port":"80",
"data_port":"8888",
"active_mode":"AP (4A:3F:DA4F2C:DB)",
"ap_ssid":"ESP3D",
"ssid_visible":"Yes",
"ssid_authentication":"WPA",
"ssid_max_connections":"4",
"ssid_dhcp":"Started",
"ip":"192.168.0.1",
"gw":"192.168.0.1",
"msk":"255.255.255.0",
"connected_clients":[{"bssid":"00:08:CA:33:C7:B4",
"ip":"192.168.0.100"}],
"disabled_mode":"STA (48:3F:DA:4F:2C:DB) is disabled",
"captive_portal":"Enabled",
"ssdp":"Enabled",
"netbios":"Disabled",
"mdns":"Enabled",
"web_update":"Enabled",
"pin recovery":"Disabled",
"autentication":"Disabled",
"target_fw":"Marlin",
"M117_output":"Enabled",
"Serial_output":"Enabled",
"Websocket_output":"Enabled",
"TCP_output":"Enabled",
"fw":"2.1.1.b8 ESP8266/8586"}

luc-github, 2020-12-01:
ok you are in AP mode
so it means if you are in captive portal you cannot upload
the system block some javascript API
enter 192.168.0.1 in your browser to not use captive portal

lidas1234, 2020-12-01:
tried but through the browser does not let me in, what can I do

> Il giorno 01 dic 2020, alle ore 10:41, Luc <notifications@github.com> ha scritto:
> 
> ok you are in AP mode
> so it means if you are in captive portal you cannot upload
> the system block some javascript API
> enter 192.168.0.1 in your browser to not use captive portal
> 
>

luc-github, 2020-12-01:
[image]

luc-github, 2020-12-01:
do a screen of all page pls

lidas1234, 2020-12-01:
i discovered one thing i don't copy because, my router changes static ip set on printer, this is it now

> Il giorno 01 dic 2020, alle ore 10:57, Lidiano Marcassa <marcassalidiano@gmail.com> ha scritto:
> 
> 
> <Schermata 2020-12-01 alle 10.52.52.png><Schermata 2020-12-01 alle 10.53.49.png><Schermata 2020-12-01 alle 10.55.49.png><Schermata 2020-12-01 alle 10.56.09.png>
>> Il giorno 01 dic 2020, alle ore 10:50, Luc <notifications@github.com <mailto:notifications@github.com>> ha scritto:
>> 
>> do a screen of all page pls
>

lidas1234, 2020-12-01:
but now it gives me this error, "Use only names and files in 8.3 format" what does it mean what should I do, thanks

> Il giorno 01 dic 2020, alle ore 11:18, Lidiano Marcassa <marcassalidiano@gmail.com> ha scritto:
> 
> i discovered one thing i don't copy because, my router changes static ip set on printer, this is it now
> 
> <Schermata 2020-12-01 alle 11.16.46.png>
> 
> 
> 
>> Il giorno 01 dic 2020, alle ore 10:57, Lidiano Marcassa <marcassalidiano@gmail.com <mailto:marcassalidiano@gmail.com>> ha scritto:
>> 
>> 
>> <Schermata 2020-12-01 alle 10.52.52.png><Schermata 2020-12-01 alle 10.53.49.png><Schermata 2020-12-01 alle 10.55.49.png><Schermata 2020-12-01 alle 10.56.09.png>
>>> Il giorno 01 dic 2020, alle ore 10:50, Luc <notifications@github.com <mailto:notifications@github.com>> ha scritto:
>>> 
>>> do a screen of all page pls
>> 
>

luc-github, 2020-12-01:
you can only use a file name with 8 chars + 3 chars extensions and no spaces, long names are not accepted for upload - it is Marlin `feature`
so `myfile.gco` is ok  = 6 + 3
as well as `myfile01.gco` is ok  = 8 + 3
but `my file name.gcode` is **NOT** ok because not 8.3 and have space

lidas1234, 2020-12-01:
done put the name as you said, it started to load but after a while the printer rings and this error appears on the display

Inviato da iPhone

luc-github, 2020-12-01:
email do not link pictures - you need to use github to share picture
Cannot see it

lidas1234, 2020-12-01:
[image]

closed 2020-12-01 by lidas1234

luc-github, 2020-12-01:
be sure you use latest FW for your TFT if not you cannot upload - TFT queries will poluate the upload - and make it failed

lidas1234, 2020-12-01:
that i know i have the latest software

> Il giorno 01 dic 2020, alle ore 12:04, Luc <notifications@github.com> ha scritto:
> 
> be sure you use latest FW for your TFT if not you cannot upload - TFT queries will poluate the upload - and make it failed

luc-github, 2020-12-01:
what give `M115 TFT` from webUI ?

lidas1234, 2020-12-01:
M115 TFT
FIRMWARE_NAME: BIGTREE_TFT35_V3.0.26.x SOURCE_CODE_URL:https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware
Cap:HOTEND_NUM:1
Cap:EXTRUDER_NUM:1
Cap:FAN_NUM:1
Cap:FAN_CTRL_NUM:0
ok
> 
> what give M115 TFT from webUI ?

luc-github, 2020-12-01:
so yes FW looks ok 
does the upload failed ? or just the error popup on TFT ?

lidas1234, 2020-12-01:
Failed to load error popup on TFT

> Il giorno 01 dic 2020, alle ore 13:35, Luc <notifications@github.com> ha scritto:
> 
> so yes FW looks ok
> does the upload failed ? or just the error popup on TFT ?

lidas1234, 2020-12-02:
no other idea to solve the problem

Inviato da iPad

luc-github, 2020-12-02:
>Failed to load error popup on TFT

You seems to set CPU freq to 80Mhz when readme metion to use 160Mhz
You still did not shared WebUI version - so may be need webUI UI update
You shared TFT error but not WebUI error ? Does it means upload is ok ?

lidas1234, 2020-12-02:
So I tried again today, and I also post the error I have on WebUI
[image]

luc-github, 2020-12-02:
yes it says ESP_ERROR_FILE_WRITE because it failed to get proper ack when sending - I suspect the transfer get some polution from either printer or tft looking at the terminal behind the error box

lidas1234, 2020-12-02:
What other information do you need
the webUI version is the latest

lidas1234, 2020-12-02:
I disabled the automatic temperature, tried to upload the file but it gives me the same error

lidas1234, 2020-12-03:
do you have any other ideas because I don't know what to do

luc-github, 2020-12-03:
No I do not sorry

luc-github, 2020-12-03:
the only thing I have in mind is :
* be sure the gcode file is valid text gcode file
* try to upload with pronterface and check if result file is correct