luc-github/ESP3D #545: [BUG]<Enter comprehensive title>

treepleks, 2020-12-20:
In "Smoothieware target" mode, with ESP3D running one a Sonoff device that controls the power of the 3D printer, asking for the printer configuration turns the printer off, making the configuration data useless (it cannot be saved onto the printer).

**To Reproduce**
Steps to reproduce the behavior:
1.  Turn the Sonoff ON using [ESP201]P12 V1
2. Click on ESP3D "Printer tab"
3. Clink on the Printer configuration "button"
4. The printer turns off (the configuration cannot be saved)

**Expected behavior**
Asking for the printer configuration should not change the GPIOs (especially GPIO 12).

**ESP3D Firmware:**
No firmware change, the issue has been there for quite a time.
SDK: 2.2.1(cfd48f3)
FW version: 2.1.1.b7 ESP8266/8586
WebUI version: 2.1b68

**Target Firmware:**
 - Smoothieware
 - Smoothie   Build version master-1089498NOMSD, Build date Dec 15 2016 12:45:49
 
**Board used (please complete the following information):**
 - MCU: ESP8266 
 - Name: Plain Sonoff device
 - Flash size:1M 
 - Available Size for update: 502.17 KB(Ok)
 - Available Size for SPIFFS: 125.25 KB

luc-github, 2020-12-20:
askin configuration does not change any gpio, it just send `cat /sd/config` or `cat /sd/config.txt` if first is not valid
https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/www/js/config.js#L33-L36

So your issue is not that simple 

Please type  in web UI monitor `cat /sd/config` to see what happen

treepleks, 2020-12-21:
Sorry for the lack of 'issue' title Luc.

The 'cat /sd/config' also does turn the printer off. I'm baffled ! Did you test this on a Sonoff ?

luc-github, 2020-12-22:
what about rename a copy of it like data.bk and do 'cat /sd/data.bk' to see if it also has same behavior

luc-github, 2020-12-22:
well I found the problem I think
```
switch.psu.output_on_command                echo_[ESP201]P12_V1
switch.psu.output_off_command               echo_[ESP201]P12_V0
```

The esp3D see this  command and execute it

luc-github, 2020-12-22:
can you remove these lines - not comment - really remove - to verify it is the root cause ?

treepleks, 2020-12-22:
It is indeed the root cause. 

1) removing these 2 lines plus another one later that also includes the same ESP3D command (to turn off the printer from Smoothie LCD menu) effectively removes the problem.

2) changing the filename has no effect.

So ESP3D interprets the flow and this surprises me. I thought it would be just listening to the output and parsing it as a text flow not as a command flow. 

If I can help to change this, just tell me. I'm C/C++ capable :-)

luc-github, 2020-12-22:
Ok I will fix this today - it is easier when got the root cause ^_^

treepleks, 2020-12-22:
Great. Thanks to Smoothieware to require the underscore to emulate the white space :-)

luc-github, 2020-12-22:
please give a chance to this commit https://github.com/luc-github/ESP3D/commit/99b6261aa451ee15e799693a90375fa8aa279505

it cover the case you meet  (hopefuly)

treepleks, 2020-12-22:
Damned!
I compiled with VS Code/PlatformIO and got 
`-rw-rw-r-- 1 tom tom 499840 déc.  22 16:25 firmware.bin
`
I'm supposed to have an "Available Size for update: 502.17 KB(Ok)"

But the OTA update says: Error 5,     Upload rejected. Any idea of what this error 5 is ? Size ?

luc-github, 2020-12-22:
yes size #define ESP_ERROR_NOT_ENOUGH_SPACE 5

I think there is a problem is size calculation 
1MB flash 

FW ~500KB SPIFFS = 125KB  - I do not see how update could be  more than 375KB

luc-github, 2020-12-22:
can you do a [ESP420]plain and give the output ?

treepleks, 2020-12-22:
Strange indeed. 
```
[ESP420]plain
Chip ID: 11527620
CPU Frequency: 160Mhz
Free memory: 37.26 KB
SDK: 2.2.1(cfd48f3)
Flash Size: 1.00 MB
Available Size for update: 502.17 KB(Ok)
Available Size for SPIFFS: 125.25 KB
Baud rate: 115200
Sleep mode: None
Channel: 1
Phy Mode: 11g
Web port: 80
Data port: 8888
Hostname: KosselMetal
Active Mode: STA (84:F3:EB:AF:E5:C4)
Connected to: Schiex-Home2
Signal: 100%
IP Mode: DHCP
IP: 192.168.8.199
Gateway: 192.168.8.1
Mask: 255.255.255.0
DNS: 192.168.8.1
Disabled Mode: AP (86:F3:EB:AF:E5:C4)
Captive portal: Disabled
SSDP: Disabled
NetBios: Disabled
mDNS: Enabled
Web Update: Enabled
Pin Recovery: Disabled
Authentication: Disabled
Target Firmware: Smoothieware
M117 output: Enabled
Serial output: Enabled
Web socket  output: Enabled
TCP output: Enabled
FW version: 2.1.1.b7 ESP8266/8586
```

treepleks, 2020-12-22:
Would there be a minimal firmware stub that I could flash first and then OTA for real?

luc-github, 2020-12-22:
yes https://github.com/luc-github/ESP3D/discussions/331

luc-github, 2020-12-22:
I do not know why your calculation side is wrong on my side using Platformio flash it is correct - size is not enough
```
[ESP420]plain
Chip ID: 14152657
CPU Frequency: 160Mhz
Free memory: 37.68 KB
SDK: 2.2.1(cfd48f3)
Flash Size: 1.00 MB
Available Size for update: 306.67 KB(Not enough)
Available Size for SPIFFS: 228.20 KB
Baud rate: 115200
Sleep mode: Modem
Channel: 1
Phy Mode: 11g
Web port: 80
Data port: 8888
Hostname: MYESP01
Active Mode: STA (18:FE:34:D7:F3:D1)
Connected to: WIFI_OFFICE_B2G
Signal: 66%
IP Mode: DHCP
IP: 192.168.1.73
Gateway: 192.168.1.1
Mask: 255.255.255.0
DNS: 192.168.1.1
Disabled Mode: AP (1A:FE:34:D7:F3:D1)
Captive portal: Enabled
SSDP: Enabled
NetBios: Disabled
mDNS: Enabled
Web Update: Enabled
Pin Recovery: Disabled
Authentication: Disabled
Target Firmware: Smoothieware
M117 output: Enabled
Notifications: Disabled
Serial output: Enabled
Web socket  output: Enabled
TCP output: Enabled
FW version: 2.1.1.b9 ESP8266/8586
```

treepleks, 2020-12-22:
Something messy happens. 
I compiled ESP3D with the suitable line uncommented, got a small firmware.bin, renamed it to update.bin, gzip'd it, sent it for OTA update via the WebUI. It accepted it, then... after the 40", nothing changes. It still claims to be in 2.2.2b7 (not b9) when I ask [ESP420]plain. I nevertheless tried to flash again the full firmware, it was still rejected. I gzip'd it and it got accepted but, again, no change in ESP420. Stll b7.

I checked the dates of the files I sent for OTA and they are the ones I just compiled.

Could my Flash memory be dead ? I didn't flash it a lot (AFAICR).

luc-github, 2020-12-22:
not sure where the issue come from ... the upload is fine but may be the gz deflating failed 
I am checking the arduino IDE which mention : FS144K / OTA 430KB 
So the calculation is may be wrong in ESP3D but it should be rejected if not enough space

treepleks, 2020-12-22:
Well. Let's pause here. I will try to find time to reflash the ESP in the usual way, not through OTA. I hate the Sonoff for this, the pin header is not easy to access in a reliable way. Possibly tomorrow... who knows :-)

luc-github, 2020-12-22:
Yes it become also difficult to update 1MB Flash using OTA - I am checking now as I am not even able to update at all on esp01

luc-github, 2020-12-22:
Ok I had to read my notes and read my own docs properly
1 - the current PIO use 2.5.2 esp8266 arduino core (PIO 2.2.0)
In 2.5.2 I had issues with 144KB - FORMAT was crashing  for example, so I moved to 256KB for SPIFFS

2 - The gzip is supported since 2.7.1 as mentionned in thread https://github.com/luc-github/ESP3D/discussions/331#discussioncomment-191668, so using it with previous core 2.5.2 which is default in current PIO won't work

3 - The 144KB SPIFFS is working well in ESP8266 core 2.7.4, no more crash when using it

So I have added a new entry in platformIO.ini for ESP8266 core 2.7.4 (PIO 2.6.2) for ESP01-160Mhz

Conclusion:  sorry you have to flash with latest Platformio.ini (so no OTA to update) your sonoff and choose 2.7.4 - after updater only will work as well as gz method, you may change the SPIFFS to 144KB to get more space for OTA

2.7.4 does not like too much SPIFFS so you will get bunch of warnings telling you SPIFFS is deprecated - just ignore them

I have tested on my ESP01 and it update as expected.

Took some time to dig this issue as I do not use PlatformIO for 2.1 and I do not use OTA with ESP01 as I know it take some effort, so I usually flash directly.

luc-github, 2021-01-01:
@treepleks can issue be closed?

luc-github, 2021-01-03:
I guess yes another commit complete this issue https://github.com/luc-github/ESP3D/commit/f1b31c68641667e3d72a4194fc8f92357b85bb61

closed 2021-01-03 by luc-github