luc-github/ESP3D #550: [FEATURE REQUEST] On ESP32-S2, support usb-to-serial

lsr1, 2021-01-01:
[Many thanks for your great project! This is not a "request", but merely a suggestion; I'm not a 100 % sure if this is even feasible.]

TL;DR: ESP32-S2 supports USB; many printer-boards do not have dedicated RX/TX pins but only an USB port that acts as UART port (e.g. with an FTDI chip). If ESP3D could support the appropriate driver, this would be awesome. This depends on #490. 

Long Version:
There exist 3d printer boards that do not have RX/TX pins for UART; but nearly all boards have a USB port that appears as UART device to a host. Since the ESP32 does support USB natively, it would be awesome if ESP3D on its ESP32-S2 version could support the (one, or two, or selectible, or most-common) USB-to-UART chipsets. This would enable a user to not have to e.g. use a "whole  Raspberry Pi" next to the printer. 

I have researched if somebody built a "wifi-to-usb-to-UART" bridge already for esp32-s2, but came up empty (my Google skills might be lacking). 

Plz feel free to delete/remove this suggestion for any reason.

luc-github: If you find this intriguing and it would help you, plz PM me - I'd be happy to order an ESP32-S2 dev board on Ali and have it shipped directly to you (e.g. https://www.aliexpress.com/item/4001205160373.html )

luc-github, 2021-01-01:
the S2 support is already planned https://github.com/luc-github/ESP3D/issues/490 but my understanding is : usb host is for Disk/device only but I may be wrong

Anyway I will start support when the master branch will support it - currently it is on only in test branch , I have  already a couple of boards ready for it 

For the USB support there is another solution discussed here : https://github.com/luc-github/ESP3D/issues/424 -  O have ordered 2 boards from hobbytronics but did not really tested yet

lsr1, 2021-01-01:
Many thanks for your reply! I did read all subjects of feature requests, but the topic of #424 did not lend itself to what I was looking for. I'll try to mark this/my issue as duplicate/closed.

luc-github, 2021-01-01:
ok my mistake the USB / serial seems supported : https://github.com/espressif/arduino-esp32/blob/esp32s2/libraries/USB/examples/USBSerial/USBSerial.ino

luc-github, 2021-01-03:
So I guess issue can be closed as duplicate as https://github.com/luc-github/ESP3D/issues/490

closed 2021-01-03 by luc-github