luc-github/ESP3D #551: [Question] NodeMCU ESP32 Dev / CR-10 v.3 (Marlin 2.0) Do i have the correct setup?

jaddel, 2021-01-03:
[image]
[image]

I couldn't mange to control the printer via the webinterface of ESP3D.

Is the following setup correct? 

I connected the NodeMCU as follows:

Printer FW Marlin 2.0 = NodeMCU ESP3D
GND = GND
RX = TX
TX = RX
5V = 5V

Am I missing any essential steps to get the setup working?

luc-github, 2021-01-04:
be sure Baud rate is same between esp3d and board
also you may try to invert rx/tx 
be sure usb is not connected on nodemcu neither board

I see it special connector for tft if so may be the port is not enabled in Marlin ?

luc-github, 2021-01-07:
@jaddel any update?

jaddel, 2021-01-07:
Hi, thanks for the suggestion. I'm currently busy and will look into it again when my silent fans arrive so i don't need to open the PSU twice.

luc-github, 2021-01-19:
I move topic to discussion instead of bug tracker

luc-github, 2021-01-31:
As transfer to discussion seems still not back I close issue

closed 2021-01-31 by luc-github