luc-github/ESP3D #558: [Question] ESP32-CAM

DangerD1024, 2021-01-13:
How can i access camera after flashing firmware?

luc-github, 2021-01-13:
use the webUI 3.0 check https://github.com/luc-github/ESP3D-WEBUI/issues/123

DangerD1024, 2021-01-13:
Sorry, do you have compiled index.html.gz ?
[image]

luc-github, 2021-01-13:
yes as writen in webUI read me https://github.com/luc-github/ESP3D-WEBUI/tree/3.0/dist/printer/production

DangerD1024, 2021-01-13:
Uploaded version 3.0 with uncomented camera
[image]
[image]
Still can't find it =)

luc-github, 2021-01-13:
did you check the link and the comment above https://github.com/luc-github/ESP3D/issues/558#issuecomment-759550894 ? there are pictures

luc-github, 2021-01-13:
@DangerD1024 did you solved your issue ?

DangerD1024, 2021-01-13:
Works, thank you. Would be great to update wiki, so dummys like me will not ask that again =)

luc-github, 2021-01-13:
feel free to contribute ^_^

closed 2021-01-14 by luc-github