luc-github/ESP3D #559: [BUG]<Enter comprehensive title>

marcinbauer85, 2021-01-14:
**Describe the bug**
Cannot connect to first time AP.
Hardware:
- ESP07 (1MB)
- USB to ESP07 adapter (https://cutt.ly/LjWg5wV)
- Macbook Pro 2019
Software:
- VSCode (fresh install)
- PlatformIO (fresh install)
- ESP3D (2.1.1)

**To Reproduce**
1. Opened latest release of ESP3D in Platformio/VSCode
2. Edited platformio.ini to:
`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
src_dir     = esp3d
build_dir   = .pioenvs
lib_dir     = libraries
libdeps_dir = .piolibdeps
data_dir = esp3d/data
default_envs = esp8266

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
;theboard
board = esp32dev
framework = arduino
monitor_speed = 115200
; set frequency to 240MHz
board_build.f_cpu = 240000000L
; set frequency to 80MHz
board_build.f_flash = 80000000L
board_build.flash_mode = qio
; None
build_flags = -DCORE_DEBUG_LEVEL=0
board_build.partitions = min_spiffs.csv
upload_speed = 921600
lib_ignore = 
    ESPAsyncTCP

[env:esp8266]
;esp8266 core 2.5.2
platform = espressif8266@2.2.0
platform_packages      = toolchain-xtensa@2.40802.200502
board = **esp07**
framework = arduino
monitor_speed = 115200
; set frequency to 160MHz
board_build.f_cpu = 160000000L
; set frequency to 40MHz
board_build.f_flash = 40000000L
board_build.flash_mode = dio
upload_resetmethod = ck
build_flags = 
    -Wl,-Teagle.flash.4m2m.ld
    -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
    -DNONOSDK221=1
    -DNDEBUG
    -DVTABLES_IN_FLASH
upload_speed = 115200
lib_ignore = 
    AsyncTCP
    ESP32NetBIOS
    ESP32SSPD
`
3. Click build (success).
4. Connected ESP07 via the usb converter set to PROG
5. Click Upload (success).
6. Unplung USP07, change PROG -> UART
7. Search for new wifi AP, came up as: "ESP-EAD279"
8. On trying to connect I get an error on 
**Expected behavior**
Connect to AP and continue to configure

**Screenshots**
[https://files.fm/thumb_show.php?i=6m2rctdxb](url)

**Target Firmware:**
N/A

**Board used (please complete the following information):**
- Generic ESP07 (1MB)

**Additional context**
Log from build:
https://hastebin.com/muxafiqeke.yaml
Log from flash:
https://hastebin.com/vaqapucohe.yaml

luc-github, 2021-01-14:
you use the wrong settings - use the ESP01 as it is without changing anything
you used a 4M sized config on your ESP07 that has 1M

marcinbauer85, 2021-01-14:
Sorry Im confused. What should I change?
`board = esp01' ?

luc-github, 2021-01-14:
I wrote : do not change anything
use the esp01 environment : esp01s_160mhz-2.7.4

do not change anything

marcinbauer85, 2021-01-14:
Ah ok! Sorry I downloaded the latest soruce from the releases 2.1.1, which didn't have this env you were talking about. I downloaded the main one and after clean/build/flash it all works!

Thank you!

luc-github, 2021-01-14:
great - have fun ^_^

closed 2021-01-14 by luc-github