luc-github/ESP3D #560: [BUG] Impossible to upload gco file on SD card.

RegisVy, 2021-01-14:
**Describe the bug**
Impossible to upload gco file on SD card.

**To Reproduce**
Steps to reproduce the behavior:
1. in the web UI in the SD File interface, clic on "Upload"
2. Choose a file (i.e. Test1.gco)
3. The status bar indicates 100% uploaded very quickly and the printer bed and nozzle start to heat
4. When the target temperature is reach, "upload failed" and this message appears
"Error:Line Number is not Last Line Number+1, Last Line: 8
Resend: 9
ok
Deletion failed (read-only), File: Test1.gco.
ok"
And "Error(3) File write failed" pop up message appears.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ESP3D Firmware:**
- ESP core version: [2.1.1]
- Version:[2.1.1.b8 ESP8266/8586]
- Web UI Version:[2.1b57]  
- Wifi mode:[STA]  
- Flash method: [PlatformIO]   

**Target Firmware:**
 - Name: Marlin
 - Version: Marlin 2.0.7.2
    
**Board used (please complete the following information):**
 - MCU: [ESP32] 
 - Name:[e.g. NodeMCU2s] 
 - Flash size: [1.00 MB]
In platformio.ini : 
build_flags = 
    -Wl,-Teagle.flash.1m256.ld

**Additional context**
- The Test1.gco file is very simple (1KB) ;
- I am able to move the nozzle and the bed through all axes from the web UI ;
- I am able to command the temperature from the WebUI.
- I am a noob/beginner  sorry for that.

[TestFile_ErrorMsgScreenShot.zip](https://github.com/luc-github/ESP3D/files/5816795/TestFile_ErrorMsgScreenShot.zip)
[Configuration.zip](https://github.com/luc-github/ESP3D/files/5816828/Configuration.zip)

luc-github, 2021-01-14:
do you have autotemp in Marlin enabled?

RegisVy, 2021-01-15:
Sorry, I forget to say hello in my first post,

Yes, the AUTOTEMP is enable in the configuration_adv.h of Marlin, here is a screenshot :
[image]

luc-github, 2021-01-15:
so that the problem - you must not have any message on going during upload - Marlin do not stop autosend when using M28/M29 - disable aauto send and check no message is ongoing to ESP by checking the terminal in web UI  - if any message take a screen
if no message then you can do upload

RegisVy, 2021-01-15:
Thank you for you answer, I tested the following : M155 S0 to desable temperature autoreport : 

Marlin doc : 
[image]

I typed M155 S0 : 
[image]

But, at the upload, I still have the same error message : 
[image]

And the same information in the web ui terminal : 
[image]

I am suprised to still see some information about temperature in the terminal since I sent the command M155 S0.

Question : 
1) Did you want I typed this command (M155 S0) or you were thinking about something else?
2) Do you want me to desable in marlin firmware directly the AUTO_REPORT_TEMPERATURES ?
[image]

Thank you in advance

luc-github, 2021-01-15:
are you uploading when you are heating ?
I see the temperature message from Marlin

RegisVy, 2021-01-15:
At the moment I start to upload, the printer starts to heat at the temperature defined in the Test1.gco .
It behavs like if it start to run the .gco file before the file finish to upload (it is exactly this behavior, if I change the temperature defined in the file and try to upload, the target temperature will be update and the printer will start to heat even if the file is not uploaded)

luc-github, 2021-01-15:
it means the command M28 is  not taken in account and the fw take each command as a command instead of save to sd

can you enter the following commands in webui terminal and give me the output :
````
M20
M28 test1.gco
M117 nothing
M29
```

RegisVy, 2021-01-15:
Ok, 

M20
Begin file list
PIN.GCO 552870
XYZCAL.GCO 334725
LEAN_XY.G 2144156
LEAN_XY.GCO 2144156
TEMPPC2.GCO 4658543
TEMPPC3.GCO 4658543
TEMPPC4.GCO 4658518
TEMPPC1.GCO 4668324
TEMPPC5.GCO 5023636
BGPLA.GCO 14563903
BGPC.GCO 14641945
End file list
ok

M28 Test1.gco
echo:Now fresh file: Test1.gco
open failed, File: Test1.gco.
ok

M117 nothing
ok

M29
ok

luc-github, 2021-01-15:
as you can see Marlin cannot open the file - so upload is failing 
check if your SD is not on read only - and be sure it is 4GB SD formated with https://www.sdcard.org/downloads/formatter/eula_windows/index.html

RegisVy, 2021-01-15:
Ok I will try this and let you know, thank you.

RegisVy, 2021-01-15:
How can I check if the SD card is in read only mode? because when I connect the sd card to my pc I am able to add file with no issue...

luc-github, 2021-01-15:
if you can add file so it means it is not in read only 😸

RegisVy, 2021-01-15:
Ok good, so I will try to format the SD card with the link you sent me.

And here is a very short presentation with screenshots I just took to be sure you have all the informations you need.
[Behavior.pdf](https://github.com/luc-github/ESP3D/files/5822459/Behavior.pdf)

RegisVy, 2021-01-15:
It is exactly the same behavior after formating the SD card :(

luc-github, 2021-01-15:
well as you can see it is Marlin side - not EP3D 
what size is the SD card ?

RegisVy, 2021-01-15:
Yes, I guess so. It is an 8GB
[image]

luc-github, 2021-01-15:
try overwrite format - not quick format

RegisVy, 2021-01-15:
I am doing the overwrite.

Ok but, by just typing this command, does the system find were the test1.gco code is located on my laptop?

RegisVy, 2021-01-15:
It is still the same with the overwrite option

luc-github, 2021-01-15:
well I have no idea sorry - just change SD

RegisVy, 2021-01-15:
no problem, so from your point of view, it doesn't come from ESP3D, nether from Marlin but it comes from the SD card directly? 

With another SD card I have a different issue : 
[image]

RegisVy, 2021-01-15:
I confirm the instructions I give to the printer are working (ie set the X position) so the connection is OK

luc-github, 2021-01-15:
M28 test1.gco failing is SD card issue - Marlin support it 
The `No connection` is another issue and without details it is hard to answer - but it means esp board do not communicate with webUI so it is different than SD access

RegisVy, 2021-01-15:
Ok thank you, I will ask the Marlin support (I guess there is a Marlin support), thank you for your time. If they find the issue, Iwill update this post.

RegisVy, 2021-01-15:
I just found the issue, in Marlin installation, in the configuration_adv.h, the define SDCARD_READONLY was not commented : 
[image]

Thank you for you help!!

luc-github, 2021-01-15:
I did  not know this option - thank you for sharing 
I guess issue can be closed then ^_^

closed 2021-01-15 by luc-github