luc-github/ESP3D #561: The index.html upload rejected with the latest develoption esp3d(v3.0.0.a84)

yu3000, 2021-01-16:
my board is nodemcu-32 v1.2(2768)

The esp3d release version v2.0.1 is ok for me,but the latest develoption esp3d has some problems to upload index.html.tar to board.

The hints is:
Update failed(5) : upload rejected.

Expect your help,thanks very much.

luc-github, 2021-01-16:
1 - it does not support tar files
2 - Error 5 means not enough space - you do not have enough space on FS to upload

yu3000, 2021-01-16:
thank you very much

luc-github, 2021-01-16:
I close issue then

closed 2021-01-16 by luc-github

yu3000, 2021-01-16:
This is the output from 192.168.0.1:
Status: Upload failed, / does not exists!  |  Total space: 1023.69 MB  |  Used space: 1023.69 MB  |  Occupation: 
 2%

This is the output from ardunio:

Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: e8:68:e7:22:a6:b4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 2259.9 kbit/s)...
Hash of data verified.
Compressed 15856 bytes to 10276...
Wrote 15856 bytes (10276 compressed) at 0x00001000 in 0.1 seconds (effective 862.9 kbit/s)...
Hash of data verified.
Compressed 1128832 bytes to 645104...
Wrote 1128832 bytes (645104 compressed) at 0x00010000 in 11.1 seconds (effective 810.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 682.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

I don't know if the board is really no enough space?

luc-github, 2021-01-16:
what board is it ?

yu3000, 2021-01-16:
nodemcu-32 v1.2 
ESP32-S

luc-github, 2021-01-16:
what settings do you use in IDE to flash ?

yu3000, 2021-01-16:
ESP32-Ardunio->nodemcu-32S

luc-github, 2021-01-16:
all settings please

yu3000, 2021-01-16:
upload speed  921600
flash frequency 80MHZ

luc-github, 2021-01-16:
take screenshoot of all settings in ide like this
[image]

luc-github, 2021-01-16:
and if you use 80HZ which not correct setting so I suspect you also use another wrong settings

yu3000, 2021-01-16:
sorry，I can't see the image,but I will attach a screenshoot

yu3000, 2021-01-16:
[image]

yu3000, 2021-01-16:
[image]

yu3000, 2021-01-16:
For the nodemcu-32S,it's has only two choice ,80 or 40 MHZ,I will change to 40MHZ to try it

luc-github, 2021-01-16:
what ? no this is not the Flash Frequency the problem

yu3000, 2021-01-16:
Use the same settings,it's ok with the ESP3D 2.0

luc-github, 2021-01-16:
- use generic esp8266 board not NodeMCU
- CPU 160MHz
- Flash 4M with 2M for FS
- Flash Mode DOUT

Do not change others default settings but port

[image]

luc-github, 2021-01-16:
I just realize you are using ESP32 board ?

yu3000, 2021-01-16:
ok,I will try your given settings,thanks,
wait my feedback:)

luc-github, 2021-01-16:
my settings are for nodeMCU esp8266 your log show you are compiling for ESP32 
what is your board exactly ?

yu3000, 2021-01-16:
Yes, It's ESP32 board,so could I use the setttings of 8266

luc-github, 2021-01-16:
no

yu3000, 2021-01-16:
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme
None
MAC: e8:68:e7:22:a6:b4
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: e8:68:e7:22:a6:b4
Hard resetting via RTS pin...

yu3000, 2021-01-16:
so the problem is not coused by the settings?

luc-github, 2021-01-16:
yes could, use these settings as I do not know what nodeMCU use as FS and do not want to check the board.txt

[image]