luc-github/ESP3D #562: File transfer time

Legsmaniac, 2021-01-17:
Currently using 2.1.x and I tried uploading a .gco file from PC to Printer for the first time yesterday.
Is it correct that transfers aren't particularly fast?
The file in question was only a little over 1Mb in size yet took some 20 minutes to upload.
It be quicker to get off my fat .... take the sd card out and copy direct.
WiFi signal is -70dBm so pretty good really.

Is this correct?  
Just wanted to check in case I'm doing something wrong.

luc-github, 2021-01-17:
https://github.com/luc-github/ESP3D/discussions/535

Legsmaniac, 2021-01-17:
> please be sure you followed template or your issue may be dismissed. if you deleted the template.....

Thank you for that.  Apologies, I didn't know there was any template for the Questions request.
And no, I didn't delete any template, nothing appeared when I clicked to ask a question, the box was blank.
Hence I didn't know there should be a template.

Thank you for the link, off to peruse.........

luc-github, 2021-01-17:
I know there is bug in github when you fill bug and did not login in github already - it won't display the template - no worry
I close question as anwered

closed 2021-01-17 by luc-github

ETE-Design, 2021-01-18:
@luc-github Could MeatPack mabye be an idea to get shorter transfer times? https://github.com/prusa3d/Prusa-Firmware/pull/2955

luc-github, 2021-01-19:
Did you saw the transfer rate of 0.403 KB/Sec ? it may me 55% better than the current transfer but it is still slow like hell 
FYI the MKS protocol transfer rate is 110KB/sec https://github.com/luc-github/ESP3D/issues/523#issuecomment-756188506

You may be want to implement MKS protocol as option for all boards instead of MKS boards only - I think it only in Marlin currently