luc-github/ESP3D #566: Is there a way to keep the camera going?

Legsmaniac, 2021-01-24:
Whether I use version 2 or 3 and set up the video feed from the camera in a panel, the video stops after a short while.
I use http address to my video feed from BlueIris.  Anything else I use goes continuously.  Not so in ESP3D.
Is this a limitation or is there something I need to do differently?
Thanks.

(PS, Logged in when I started, no template again, sorry)

luc-github, 2021-01-24:
sorry can you elaborate your question ?
what camera ? how do you do link html with your camera stream ?

Legsmaniac, 2021-01-24:
It's a WiFi IP Camera connected to BlueIris software.  The BlueIris software is already set up on my server PC running all my other CCTV cameras so I thought I may as well connect the camera intended to watch over my 3D prints.
So to see any stream from BlueIris you use http://(IPAddress of BlueIris/mjpg/(name of camera)/video.mjpg and the video stream can be viewed in a browser and will continue to play until the browser is closed.
But with ESP3D, even though it's also a browser, the video stream stops after a few minutes.

luc-github, 2021-01-24:
the esp3d panel / page for camera is just an iframe - the webUI do not anything - in 2.1 you can even detach the window to be in separate place
which setup in esp3d did you do and which version are you using ?

Legsmaniac, 2021-01-25:
Apologies.
I have now discovered that it is designed to stop streaming if no activity to prevent unnecessary server load.  The only way to continue viewing is to refresh to signal activity.
So for my working setup (ESP3D 2.1.1.b10) it seems there is nothing I can do but to manually refresh.
For my test board (ESP3D 3.0) I can set a Panel auto-refresh time in the Interface Settings (so thanks for this useful feature!), settings 60 second refresh keeps the camera stream alive.

Shame there is no auto-refresh setting in 2.1.1 so looking forward to 3.0.
Unless a refresh can be added somehow?  :-)

luc-github, 2021-01-25:
but you mention you have no issue issue in web browser so should have parameter no ?

Legsmaniac, 2021-01-25:
I don't have issue in browser, or rather, I didn't.  I had forgotten I had set that tab to auto-refresh.  (Chrome Extension)
Again, my apologies.  My fault. My bad.  
Unfortunately, tab auto-refresh doesn't work with 2.1.1 as it results in the whole Connecting ESP3D thing and loads Dashboard rather than Camera.
If there's no resolution, no worries, I'm happy to wait for 3.0 to finalise.  😊

luc-github, 2021-01-25:
ok so I guess we can close topic ^_^

closed 2021-01-25 by luc-github