luc-github/ESP3D #568: [FEATURE REQUEST]USB Disk support using CH376S chip

luc-github, 2021-01-24:
Global FS support Flash and SD, now let's try to add USB Flash disk/key using SPI and CH376s Chip

This library looks very promising https://github.com/djuseeq/Ch376msc
e
Chip is using 3.3 by default but need 5V to power the USB disk
It need to test features , if work as expected then need to implement an abstraction layer for the USB FS and  add it to the global FS

luc-github, 2021-01-29:
the boards I had were not working - their pinout are different than the one showed in library, not sure what problem was
Not working
[image]
Always got `Attach flash drive first!`

I have ordered new boards and now it is working 🥇 
Working one:
[image]
Working out of the box 

[image]

So now can work on abstraction layer ^_^

luc-github, 2021-10-12:
I did not finished yet, sorry I am late on this, 
you can have a look at  https://github.com/djuseeq/Ch376msc, I use SPI wiring, esp32 use 3.3v so wiring is simple, but do not forget to add extra 5V to power the USB for the connected device

hello2afiq, 2024-10-28:
if somehow the reason why the above module doesn't work please tell us.

luc-github, 2024-10-28:
it is not yet implemented - I was busy with others tasks - that is why feature request is still open