luc-github/ESP3D #570: ESP3D No connection Error

Btoz25, 2021-01-28:
Hi everyone, i can't upload anything from SD files 

Gives this error:

[image]

MY ESP3D FW and UI versions:

[image]

Thank you for giving a time

luc-github, 2021-01-28:
Which SD ?
what board ?
What Firmware ?

luc-github, 2021-01-31:
I see no update but what I can see is :
1 -  it is a very old FW
2 - this FW seems to be the BTT FW  for esp3d so I guess it is used on TFT or SKR which is Marlin and there were several change since the fw was done - so back to 1 too old for latest Marlin 2.X

closed 2021-01-31 by luc-github