luc-github/ESP3D #571: [Question] How do I identify what firmware I have?

jschwalbe, 2021-01-30:
I have an Ender 3 Pro. Firmware is unmodified from original. Pretty new to 3D printing but I was able to get the D1 Mini wired up correctly. How do I tell which firmware I have? Thank you!

luc-github, 2021-01-30:
What Firmware you are talking about ?

jschwalbe, 2021-01-30:
I'm referring to the 3d Printer firmware, as I'm asked to select my firmware type in the ESP3D settings. i.e. Marlin?

luc-github, 2021-01-30:
Type M115 in terminal - but yet you have big chance it is Marlin if ender 3 pro

jschwalbe, 2021-01-30:
I selected Marlin, and that may be correct, but I really don't know what it is or how to find out despite googling for quite awhile. Maybe wrong search terms? 
The second reason the question is coming up is because I'm getting the 8.3 error message (https://github.com/luc-github/ESP3D/issues/520) but when I directly transfer from the SD card on my usb/laptop i have have longer filenames without any issue.

jschwalbe, 2021-01-30:
Thank you; sorry, is this the correct place to type it?
[image]

luc-github, 2021-01-30:
yes 8.3 filename is Marlin limitation

Yes it is correct place - but if no M115 answer - it means it is not really Marlin but custom FW

jschwalbe, 2021-01-30:
I'm on the discord but cannot post for 10 min, if you care to move this discussion there. Any other tips to get the actual firmware? I'm able to move the printer head around, etc.

luc-github, 2021-01-30:
what is the temperature output ?
`M105`

jschwalbe, 2021-01-30:
No output from that command. Could that mean poor soldering job?

luc-github, 2021-01-30:
you mean any command you type is doing nothing ?
But pressing job button is working ?

luc-github, 2021-01-30:
Unresponsive FW was related to printer was heating and so do not anwer to any commands until temperature was reached 

So closing now

closed 2021-01-30 by luc-github