luc-github/ESP3D #572: [FEATURE REQUEST]ESP32-cam flashlight

BisSpaeterDannnoch, 2021-01-30:
Hi, hope you are well.
It would be great if we could switch the built in flash light of the ESP32-Cam module

A nice way would be to trigger the light from the web interface  with a simple push button for the live video.
And/or to control it generally from the config.ini.

My printer is mainly working in the dark and this feature will help in a great way to see if everything is in order.
Thank you for considering it - take care.

of cause this feature request will end in a time-laps video stored of the sdcard ;-) 
so consider wisely

luc-github, 2021-01-30:
the flash light pin is also used by SD
https://github.com/luc-github/ESP3D/blob/3.0/esp3d/src/include/pins.h#L165
If you want to control the flash light and you do not need SD just change the pin from -1 to 4 
and you can control it using light parameter
`[ESP170]light=1`

BisSpaeterDannnoch, 2021-02-02:
well... thank you. i could have seen it when i studied the schematics of the ESP32-cam. 
Sorry for bothering and thank you for you really quick response. take care.

closed 2021-02-02 by BisSpaeterDannnoch