luc-github/ESP3D #575: [FEATURE REQUEST]Use better serial protocol communication

luc-github, 2021-02-01:
Current protocol between printer and ESP3D use raw serial which is ok for GCODE communication but no really efficient for file transfer
More over if some output come from printer board due to auto send it bring some trouble on file transfer which is already slow

Current MKS protocol encapsulate each command in a frame with specific header it use 2 additionnal pins for notification/ack and different header for GCODE / File transfer / Error which make coexistance easier
This protocol allow 110KB/S transfer vs 0.4KB/S M28/M29 transfer and handle GCODE even during transfer.
That said current MKS protocol miss 2 things:    
* CRC to ensure data are correct
* resend process in case of corrupted data

I think add a CRC32 should improve reliability and will only use couple of byte on each frame, and calculation should not affect too much performances.  if done when data is received
Based on this crc if any error happen then do a resend - if more than 2 resend is needed  on same packet then automaticaly ask for decreasing baudrate to do more secure transfer 

This would need to add some code in Printer FW  but will ensure communication for file transfer as well as commands 
This will be between esp3d and printer like on MKS Wifi so external progam won't be affected.

I am considering to write a library that could be used to hook the serial wifi port, but this also means need a dedicated wifi port like BTT do for board or TFT or an available UART port  
If Serial port  is same as USB one, it won't work (e.g: Mega/Ramps and aux 1),  so I need to see if can handle raw serial in same time or consider it as limitation= no support of shared solution, may add custom GCODE  to switch from one mode to another ?
In addition it need 2 additional pins available.

This would allow to remove the need of supporting the slow and unreliable M28/M29 protocol but bring more work on ESP3D part by adding a daughter library / Serial protocol accross FW  (Repetier / Marlin / Smoothieware/ TBD...)

Any feedback /comment is welcome

luc-github, 2021-02-23:
if you consider M28/M29 protocol vs currentMKS protocol 
M28/M29 upload speed: 0.4KB/s
MKS protocol upload speed: 110KB/S
So SD Write is definitly a parameter but  protocol transmission is a big one 

For transfer WebServer vs other server is also a parameter, as well a SD library used  like I have rased here : https://github.com/luc-github/ESP3D/issues/383#issuecomment-548695909- that is why I would like to do https://github.com/luc-github/ESP3D/issues/576

I disagree : pin ack is faster, read a pin state is faster than read a frame content and when you need a lot of ack it is an important parameter

Serial is easy to get noise during transfer - especially on DIY project, so it must be taken in consideration : checksum+ retry
Also I wish transfer can do binary as well text - because transfer should not handle content - just data   integrity, but keeping good performances

I do not pretend I have solution, I would prefer to not put my hand on this, because  it will be needed on several FW and so it imply some work in each on,  
last but not least: I am lazy - I would like to avoid to reinvent things 😸 , M28 binary protocol is still experimental after several years, when MKS protocol is already used for several years - That is why I think it is not bad starting point.

netzmark, 2021-05-06:
Hi @luc-github , may I know if you've tried to talk to anybody of Marlin community about the protocol improvement (meaning replacing M28 with some faster one) ?

luc-github, 2021-05-06:
@netzmark no I did not, and do not have any plan, why ?
and to clarify I do not plan to replace M28/M29 which is limited to file transfer but use another protocol for communication that include file transfer but not only.
My approach will be more an external lib that will hook second serial port in same way MKS do but without filtering answer and allowing check sum and retries which are not present in current MKS protocol

netzmark, 2021-05-06:
I've meant replacement M28/M29 protocol with another more powerful, or adding the second protocol for files upload. But It would make sense if there were some Marlin developer interested to apply your protocol there. And this is what I have asked for... because I have tried to ask there on the Discord if someone thinks about it - not answered.

luc-github, 2021-05-06:
I am not focused on Marlin - this feature is for repetier / smoothieware / etc ... any FW supporting Serial communication and even not supporting M28/M29  protocol

>I have tried to ask there on the Discord if someone thinks about it - not answered.

When I ported ESP3D for Marlin used on ESP32 board I did the approach to ask - was useless - no Marlin Dev answered, I was told that the marlin mindset is: `just push your code and let see`, that is why my code in Marlin is limited to some entry points but code is often broken so I only fix for release not bugfix, that is why ESP3DLib is limited compare to ESP3D standalone - I wanted to see the impact before spend too much time on it.

Again the protocol I propose is not focused on file upload, file upload is part of protocol not the protocol itself

netzmark, 2021-05-06:
Gotcha, but it looks like nobody there (Marlin comm) is interrested in it :-(.

At occasion, I don't want to highjack the topic so shortly only:
- "Pannucat stand alone module" (the one with ESP8266 and SD card), I understand your ESP3D can be loaded there. But is there any way to connect it to SKR? If it is - what files transfer speed is possible then? Sorry I ask you not them but I count you have played with it maybe.
- MKS wifi modul connetcted to TFT28: ESP3D can be loaded there but I understand that won't get 100kB files upload to the TFT SD card, correct?

luc-github, 2021-05-06:
I do not have the Panucatt wifi backpack standalone module yet, I have ordered some but still waiting.
Then you hook it on EXP1/EXP2  like there is a TFT with SD card - the last time it was checked the speed was around 280KB/S I think but it depend also on SD card and SPI speed

about ESP3D on MKSTFT and MKS WIFI module - I got 110KB/S for file upload - same as existing MKS firmware because I use same protocol

netzmark, 2021-05-06:
Pls let know somehow when you have the standalone tested. It looks like promising stuff.

So I understand I can upload your ESP3D to MKSWIFI or into ESP01 - both should work connected to MKSTFT (properly wired of course). Great!

luc-github, 2021-05-06:
I do not have the standalone but the others and it works well 
FYR: MKSWIFI is an ESP12 not ESP01 and it is MKS module - several people got transfer issue using another modules instead of  MKS one FYI

netzmark, 2021-05-06:
I have assumed that if ESP12 contains ESP8266 inside so it is pissible to easy use ESP-01 either...
But I will get ESP12 in the next week anyway so will play with this :-).

> I do not have the standalone but the others and it works well
You told "I have ordered some but still waiting." so I guess you will get it finally. I'm curious what the upload speed to its onboard SD it will reach.

luc-github, 2021-05-06:
MKS TFT WIfi Connector is same as ESP01 but the  pins are not matching so it won't work unless you use dupont cable and doing this you won't reach the speed of 110KB/S because of EMI issue, also ESP01 use GPIO 0 and GPIO2 which are both bootstrap pins when on ESP12 they use GPIO0 and GPIO4 for communication - I am guessing the GPIO4 is put low at start which can not be with GPIO2
Additionnaly ESP12 has 4MB flash when ESP01 only have 1MB so OTA is no more possible with ESP3D as FW start to be big

netzmark, 2021-05-06:
I have fast 8MB Winbond on my ESP01 (I have soldered it instead of original 1MB CFeon). But now I see GPIO is complicating it a bit more than I feel up to do it. I will wait for MKSWIFI so :-).

luc-github, 2021-11-20:
I have started to drop some notes https://github.com/luc-github/ESP3D/tree/3.0/to_do/CommunicationProtocol

ETE-Design, 2022-01-19:
@luc-github Will it be possible to use the "MKS Protocol" -  But on the ESP32-Cam and use lets say GPIO0 + GPIO4 like MKS use on their MKS Wifi Boards... That way is should be possible to to up the transfer speed?

luc-github, 2022-01-19:
yes but you will likely have emi noise on serial wires and the upload will failed I guess because there is no retry on fail in mks protocole.
the mks wifi module are designed to avoid emi noise , I got feedback of people doing their own and some succeed but also some failed...

ETE-Design, 2022-02-08:
@luc-github If I use "MKS Protocol" with a ESP32-Cam Board, what pin's would be used for IO1 + IO2 "GPIO0 + GPIO4"? I know it mabye will fail cause EMI Noise ;-)

luc-github, 2022-02-08:
Did you tried GPIO0 + GPIO4 ?

ETE-Design, 2022-02-08:
@luc-github No, would like to connect it correct from the start, that's why I asked.

luc-github, 2022-02-08:
I have only used MKS Wifi modules with MKS protocol, but I know ESP32Cam is pretty noisy per experience - that said I never used it for MKS protocol, GPIO 4 is the flash so it may need to change it - but at first I would try the basic : GPIO0 + GPIO4

luc-github, 2022-06-19:
I have checked the octopus and seems a good candidate for fast upload as it allow GPIO0/GPIO4 like MKS

Additionnaly as it connect ESP SPI pins to SPI_2 pins of octopus to be used by RRF - it is also a good candidate if try to port SPI drivers of RepRap to Marlin