luc-github/ESP3D #577: Input PIN for Filament Runout Sensor?

Legsmaniac, 2021-02-03:
Forgive me if this is already implemented or planned for the future but I was unable to find any info.
In the config files (either versions 2.1.1 or 3.0) particularly version 3.0, I notice there are several possible/future functions commented out, especially one for a SENSOR.  Is this able to be used for a Filament Runout Sensor?
If so, which PIN is used?  If not, could it please be considered for inclusion as a Mod?

I was looking at this idea - https://www.instructables.com/IoT-Enabled-Smart-Filament-Runout-Switch/ - but it seems sill to use yet another Arduino when I could run a wire to ESP3D.  I don't use Blynk nor do I want to but my thinking is that ESP3D could send me an alert when the Filament Runout Sense PIN goes HIGH (Or LOW, whichever) 
I'm thinking of making my own Filament Runout Sensor such as the diagram below and it would be quite easy to run a single wire from the center arm of the micro switch to an INPUT PIN on ESP3D.

Your help and advice would be most appreciated if this is already possible.
Thanks.

luc-github, 2021-02-03:
the sensor feature is to send sensor info only (Temperature, humitidy, voltage, etc..) no action on sensor just forward collected info

I am not sure about the benefit of ESP3D on filament sensor - printer FW already have such feature - so why ask ESP3D to monitor a pin that Printer FW could monitor directly ?

as action(s) to react from pin state change can differe from one user to another and need to do specific code I was more thinking on using interpreter for this but need to think more : https://github.com/luc-github/ESP3D/issues/378

Legsmaniac, 2021-02-03:
> I am not sure about the benefit of ESP3D on filament sensor - printer FW already have such feature - so why ask ESP3D to monitor a pin that Printer FW could monitor directly ?

As I stated above, to send me a Notification to my mobile phone that the filament has run out.
You are correct, printer firmware already has the ability to stop the printer and await reloading but unless I check to see, I won't be aware it has run out.  The idea is for ESP3D to notify me the printer has stopped due to filament runout so that I can refuel it ASAP to continue the print.

luc-github, 2021-02-03:
so your idea is to hook on existing pin check the pin state and according state send a notification like M600 do ?

luc-github, 2021-02-03:
do you realize that the feature is already in Marlin https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/Configuration.h#L1182

Legsmaniac, 2021-02-03:
Check the PIN state and send Notification to mobile, yes, NOT to the printer so no need for M600.  The microswitch will already be plugged into the mainboard, stopping the printer as per Marlin firmware.  The ESP3D will merely notify as per Notifications setting in the Interface (Email, Pushover, Telegram, Line) to inform of Filament Runout and that the printer has stopped and awaiting urgent attention.

luc-github, 2021-02-03:
so just change the GCODE to `M118 M600 no  more filament, please refill`
you need to enable notification in esp3d

Legsmaniac, 2021-02-03:
Ah!  Thank you.  That's one way around it I guess.
I'll give that a try.
:-)

luc-github, 2021-02-03:
I am guessing you use Marlin - but other FW should have a similar way

Legsmaniac, 2021-02-03:
> I am guessing you use Marlin

Correct.  Ender 3 V2 with Creality 4.2.7 and latest Marlin from AU site.
Thanks.

luc-github, 2021-02-04:
So I guess issue can be closed ?

closed 2021-02-04 by luc-github

Legsmaniac, 2021-02-08:
Going back to this again if I may?  Sorry.  Hope you can help further.
Now that I have Notifications working.  Sorted by the DNS fix.  😃👍
I have set it to Notify via email which is working fine.  
I receive an email every time I turn on the printer (and therefore ESP3D) which tells me the IP address it's connected to.
I'm fine with that.

OK, so my question is - How do I set up any further notifications I want?
The Wiki says I can add code to my printer settings code in Cura (or PrusaSlicer) which can inform me of finished print.
You suggested above.....

> #define FILAMENT_RUNOUT_SCRIPT "M600\nM118 [ESP600] no more filament, please refill"

Where does that go exactly?  And how does it work?
Sorry for the noob questions. I'm totally lost on this.  🤦‍♂️

luc-github, 2021-02-08:
in your end code of cura just add 

*  `[ESP600] print done` if printing from SD
* `M118 [ESP600] print done` if streaming gcode from any host

Legsmaniac, 2021-02-08:
Yes, I understood that, thank you.
But what about for the filament runout detection Notification?
How does that get triggered?

luc-github, 2021-02-08:
this is done by Marlin as mentioned above https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/Configuration.h#L1182

Legsmaniac, 2021-02-08:
Ah, gotcha!  Thanks.
Time to learn to set up and compile my own Marlin firmware now I guess.  🤔

Legsmaniac, 2021-02-10:
I can now confirm that I have successfully compiled my own firmware.  Several times in fact! lol
I started with the standware latest Marlin bugfix nightly build, then went on to Jyers fork which is excellent!
I noticed there was inclusion for ESP3D in the configuration files.  Not sure what they do exactly yet but I uncommented the lines.
I have included the M600 ESP notification and tested working.  Many thanks.

luc-github, 2021-02-10:
the ESP3DLib references in Marlin are for ESP3DLib a library for Marlin, it is used on ESP32 as main board , not for external ESP board like you do  - so uncomment them may just make the compilation failed if target is not an ESp32 board depending of Marlin version you use - some sanity check block compilation if settings are wrongly enabled