luc-github/ESP3D #578: [BUG] [ESP600] always responds with Error (500) Command Error

Legsmaniac, 2021-02-04:
**Describe the bug**
As subject title says, no matter what I try in trying to use the Notification feature, I always get Error (500) Command Error

**To Reproduce**
Steps to reproduce the behavior:
1. As per Wiki, I have set up Telegram on my mobile phone - obtained my HTTP API from BotFather, created a Channel and added MyBot to my Channel.
2. Using the Terminal, I have sent [ESP610]type=TELEGRAM T1=<bot token> T2=<@channel name>
3. On checking Setting, Features is shows.......
4. Auto Notifications - Yes
5. Notifications - Telegram
6. Token 1 - ********  (I have also tried entering my bot token API manually to be sure it's correct)
7. Token 2 - ********  (as above, I have entered my created Channel name preceded by the @ sign to be sure)
8. Setting - Enter Value (which means it's blank of course)
9. In Terminal, I enter [ESP610], Terminal responds with TELEGRAM
10. in Terminal I enter [ESP600}Hello, this is your ESP calling, Terminal responds with a pop up box Error (500) Command Error
11. I have tried various way of typing things differently in case there was something wrong in the WiKi with no luck.  (It IS a bit outdated after all since it took me a while to figure out setting up Telegram as the WiKi screenshots no longer match what happens in Telegram.  Guessing Telegram has upfated since the Wiki was written.)

**Expected behavior**
Well obviously I expect the command to send and the message to be received on my phone.

**Screenshots**
If there's any problems understanding anything above or if requested then sure, I'll take some screenshots.

**ESP3D Firmware:**
- ESP core version: [2.5.2]
- Version:3.0.0.a86
- Web UI Version:[e.g 2.1b14]  
- Wifi mode:STA
- Flash method: PlatformIO
- Any change done in FW - None whatsoever
 
 tips: to get ESP3D informations type `[ESP420]plain`    
      
chip id: 3221
CPU Freq: 160 Mhz
free mem: 25.59 KB
SDK: 2.2.2-dev(38a443e)
flash size: 4.00 MB
size for update: 1023.00 KB
FS type: LittleFS
FS usage: 136.00 KB/1.98 MB
baud: 115200
sleep mode: none
wifi: ON
hostname: Ender3V2
HTTP port: 80
Telnet port: 23
WebDav port: 8181
sta: ON
mac: D8:BF:C0:14:0C:95
SSID: <redacted>
signal: 62 %
phy mode: 11n
channel: 6
ip mode: static
ip: <redacted>
gw: <redacted>
msk: 255.255.255.0
DNS: (IP unset)
ap: OFF
mac: DA:BF:C0:14:0C:95
serial: ON
notification: ON(telegram)
Target Fw: marlin
FW ver: 3.0.0.a86
FW arch: ESP82XX 

**Target Firmware:**
 - Name: Marlin
 - Version bugfix-2.0.x (Jan 26 2021 10:20:22) (from AU site)
 
**Board used (please complete the following information):**
 - MCU: [ESP8266] 
 - Name:[NodeMCU12E] 
 - Flash size: [e.g. 4M: 2M/2M] (was left to Platformio so I assume default?)

Am I doing something wrong?  I have also tried setting Notifications to Email with the same result - Error (500) Command Error.

luc-github, 2021-02-04:
the error means it failed to send message because telegram rejected it - may be something changed on telegram API - I will check later

luc-github, 2021-02-04:
ok reproduced it -  I do not update Notifications settings in current session when updating from WebUI - but only when use [ESP610]
once I restart ESP3D I got the notifications (I have enabled also the auto notification)

What is weird is you said [ESP610] do not apply settings in current session - can you use webUI to set settings and restart ?

luc-github, 2021-02-04:
I have reset Notification info to nothing, restart ESP3D 
the [ESP610] is applying settings immediatly on my side
[image]

Legsmaniac, 2021-02-04:
OK, I'm obviously doing something wrong somewhere.
I did as you suggested, cleared out all Notification settings, restarted.
Then used WebUI to set it up again.

Still I get Error (500) Command Error.

😒

luc-github, 2021-02-04:
in webUI settings or web ui terminal ?

Legsmaniac, 2021-02-04:
webUI settings.
I have however tried both.
Always the same - error 500.

luc-github, 2021-02-04:
after webUI setting update you need to restart ESP3D
as you see [ESP610] in terminal is working immediatly - so definitly a setup issue with your bot I guess

luc-github, 2021-02-04:
or your home router block the ESP board queries to internet

Legsmaniac, 2021-02-04:
> or your home router block the ESP board queries to internet

Ah!  I'm inclined to think this may be the nail on the head.  Saying that, I'm using my own ASUS DSL-AC68U router and not any provided by the ISP which usually do block stuff. Never had any problems communicating to other projects with my ASUS so I'm gonna at last have a look and see what the logs say.

luc-github, 2021-02-04:
if you are confident your telegram setup is ok and provide proper parameter - I only see the firewall router as a root cause

Legsmaniac, 2021-02-04:
Router logs show nothing to indicate any problems.
I turned the Firewall off completely.

In fact, I used Virtual DMZ to expose the ESP completely to the Internet without restrictions.  Which should rule out the router altogether.

And still no joy!

luc-github, 2021-02-04:
enable the Debug in ESP3D config file, Telnet or Serial up to you, and do the [ESP600] command to see what appear in debug - it will tell you what is the problem

Legsmaniac, 2021-02-04:
It will tell ME what the problem is?  LOL
This is what appeared in Terminal.....

```
Y,ëKª¹­¹½Ý¹command: "[ESP3D][serial_service.cpp:166] process(): Got 255 chars in serial"
k ªÍ
command: "[ESP3D][serial_service.cpp:208] push2buffer(): buffer get 156 data "
ekho:Unknown command: "[ESP3D][serial_serv:208] push2buffer(): buffer get 137 data "
ech:Unknown command: "[ESP3D][serial_service.cpp:208] push2buffer(): buffer get 144 data "
```

It's all gobbledgook to me.

luc-github, 2021-02-04:
wow your serial is noisy how come ?

luc-github, 2021-02-04:
I have just noticed one thing: you use Static IP and no DNS is set so of course you cannot access to internet

Legsmaniac, 2021-02-04:
You know what?  I wondered that.
So where do I set the DNS because there isn't a box for it in settings.

luc-github, 2021-02-04:
before (long time ago...) it was using the gateway by default and never really tested since as I always use DHCP and no one raised the problem
I will add the setting to allow to add it

Legsmaniac, 2021-02-04:
> wow your serial is noisy how come ?

I have no idea.  Poor wiring?  Poor connections?  Length of wiring - unshielded?
The two (Tx and Rx) wires are thin and probably about some 18 inches from the UART chip to the ESP, under the printer to the side where the ESP is mounted in a box.

Legsmaniac, 2021-02-04:
> I always use DHCP

And I always use static addresses.  For **everything** on my Network.  So I know where they are, what they are.  Easy to keep track of.

luc-github, 2021-02-04:
until I do the fix this afternoon you can just change this line:
https://github.com/luc-github/ESP3D/blob/3.0/esp3d/src/modules/wifi/wificonfig.cpp#L198
`WiFi.config(ip, gateway,mask, gateway);`

Legsmaniac, 2021-02-04:
> WiFi.config(ip, gateway,mask, gateway);

Changed that line, recompiled, updated.
Now I just get "echo:busy: paused for user" and a counter counting upwards.
It also cannot find Marlin configurations.

I'll wait for the fix this arvo.
Thanks man,.  👍

luc-github, 2021-02-04:
you must disable DEBUG if you connect to Marlin - with debug on serial Marlin will become crazy

luc-github, 2021-02-04:
Fix is implemented - use latest 3.0 (89)- no need to update webUI

[image]

Legsmaniac, 2021-02-04:
Oh Dear...........

[image]

luc-github, 2021-02-04:
can you share your `[ESP420]plain `

Legsmaniac, 2021-02-04:
I finally got it working!!!!  

The Wiki is wrong.  I did say it was also out-of-date as the screenshots no longer match up to the correct process as Telegram must have updated since the original Wiki was created.

But the real mistake is the part.......

```
4 - Save the generate token and chatID in ESP3D, and set Telegram as notification supplier
[ESP610]type=TELEGRAM T1=<bot token> T2=<@channel name>
```

This should read.....

```
4 - Save the generate token and chatID in ESP3D, and set Telegram as notification supplier
[ESP610]type=TELEGRAM T1=<bot token> T2=<@botname>
```

It's the name of the bot we created and not the name of the channel that goes into T2.
No wonder I was confused and couldn't get it working!

Happy to say, it's sorted for me now.  Hope someone will update the Wiki to help others.
Or maybe others are just not as thick as I am.  😂

Legsmaniac, 2021-02-04:
Have now also tested Email as per Wiki.  That was also sending me an Error 500 before.

I can now confirm that Email is now working perfectly!!  😃
So the DNS fix was a necessary one for Static users like myself.
Thank you for that and thank you for all your help.

I really only wanted to use Email and not Telegram.
Only resorted to Telegram because Email wasn't working.
Happy Chappie now!  😁

luc-github, 2021-02-04:
humm I have followed  https://xabaras.medium.com/sending-a-message-to-a-telegram-channel-the-easy-way-eb0a0b32968
I use chat_id not bot id - bot is defined by token and room by chat id 
also when you create your bot it add `_bot` at the end of the name so do not know why you use botname 

If you get it working it is the key point anyway

Legsmaniac, 2021-02-04:
Really?  How strange.
It's definitely the bot name that BotFather asks you for before he gives you the HTTP API that is needed in T2 for me and not the channel name that is created after BotFather stage.

But yes, it's working for me, for sure.

By the way, the same problem exists in version 2.1.1 for Static users.  Same Error 500.
Guessing that needs a DNS too?  :-)

luc-github, 2021-02-04:
I guess issue can be closed - thank you for pointing out the missing DNS entry 😺 
I will check fix in 2.1  if necessary - 2.1 is based on 2.5.2 core - I need to check issue is there

closed 2021-02-04 by luc-github

Legsmaniac, 2021-02-04:
> I need to check issue is there

Well it is for me.

luc-github, 2021-02-04:
I am on it - but will do a simple fix: use gateway as dns - I would like to avoid new EEPROM settings on 2.1

luc-github, 2021-02-04:
I have  updated ESP3D  2.1 and ESP3DLib 1.0 accordingly

Legsmaniac, 2021-02-04:
Excellent, thank you.
I had already tried your suggested WiFi.config fix for gateway as DNS and mine is now working so the update was a necessary one.
Good job.  👍