luc-github/ESP3D #580: [Question] how to update FW on esp01s module?

davbcr, 2021-02-10:
What is your question ?
When i try to upload fw or simili UI from web interfacce upload failure because of not enough memory to store uoloaded files.
So Jack on platformIO i tried to flash with kelly compiled FW but it doesn't work anymore to connect to the serial pc com port not to frase the flash memory.
How can i get to update my module?

luc-github, 2021-02-10:
you need to use a ftdi, a serial usb device

davbcr, 2021-02-11:
I do know it,and i'm usinf an esp8266 to usb module with a button solder to put in short.

luc-github, 2021-02-11:
so you can flash it using this device,, I do not understand your question , neither who are jack and kelly you are mentioning

davbcr, 2021-02-11:
Sorry man T9 decided to change my writing and i don't realize before hosting. There are no Jack and kelly,i meant that i compiled the FW to update from 0 but the module cannot connect anymore to the PC's com port either if I press the short button while inserting it into the port. So I'm stuck with the module that cannot be either flashed,erased or updated. Is there any other way to erase it and start from beginning?

luc-github, 2021-02-11:
the upload mode mean to erase everything I am guessing you have an hardware issue may be GPIO 2 go to GND when it should not ?
or your issue is the ftdi module you are using may be

luc-github, 2021-02-12:
@davbcr any update

davbcr, 2021-02-12:
I'll let you know as soon as breadboard and Dupont wires comes to my home. I'll try to manually connect pin by pin and put the I0 to ground while connecting.

closed 2021-02-14 by luc-github