luc-github/ESP3D #582: [BUG]Authentication - Cannot log in

Legsmaniac, 2021-02-13:
**Describe the bug**
I set up my ESP3D v3.0 with `#define AUTHENTICATION_FEATURE` uncommented since I wanted password security.  And all I get is the box with the spinning circle, no log in. 

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. I installed a completely fresh instance of ESP3D v3.0 onto a NodeMCU 8266 and erased all flash contents.
2. Log into ESP3D WiFi network and go to 192.168.0.1, sure enough, you're presented with the log in box.
3. Enter User admin and Password admin, you have the files upload page.  Upload the index.html.gz and set up.
4. I set up Client access, Static IP and my router details as normal.  Then Restart ESP.
5. All I get in the spinning circle constantly, no log in box.
6. I also tried again but leaving it as AP access and still no log in.
7. It doesn't matter whether I leave the PW as admin default or set my own PW, it's still the same problem.

**Expected behavior**
I expect to be prompted to log in.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**ESP3D Firmware:**
- ESP core version: 3.0
- Version:3.0.0.a90
- Web UI Version:3.0.0.81 
- Wifi mode:Doesn't matter, I tried both STA (Client mode) and left it as AP, still the same problem. 
- Flash method: Arduino Ide
- Any change done in FW - just uncommented `#define AUTHENTICATION_FEATURE
 
 tips: to get ESP3D informations type `[ESP420]plain`    
       to get WebUI version check your menu header   
**_I can't as I cannot log in._** 

**Target Firmware:**
 - Name: Marlin
 - Version Marlin bugfix-2.0.x 
 
 tip: to get firmware info type `M115` in console    

**Board used (please complete the following information):**
 - MCU: 8266
 - Name:NodeMCU12f
 - Flash size: 4M: 2M/2M

**Additional context**
Nothing more to add, hope it can be sorted. Thanks.

luc-github, 2021-02-13:
can you login when in `http://<yourIP>?forcefallback=yes`

Legsmaniac, 2021-02-13:
> can you login when in `http://<yourIP>?forcefallback=yes`

Yes, that gives me the log in box........

[image]

I then have to click the Go to ESP3D Interface button as I get .......

[image]

And finally, I'm in!

[image]

luc-github, 2021-02-13:
Ok I have reproduced bug and I found typo and also a bug in processing authentication, I am on it now

Also for futur debug :
* click on version number give all informations
* also FW 90 has now a console - click on it and it will expand so you can use it to  type commands

luc-github, 2021-02-13:
it is fixed - typo and side effect of an incomplete rewrite

luc-github, 2021-02-13:
Just update the latest webui

luc-github, 2021-02-13:
I move issue to webUI