luc-github/ESP3D #583: The index.html.gz file upload rejected ( code 5)

hyzx86, 2021-02-13:
192.168.0.1 page to Upload index.html.gz Upload Rejected!
The problem is similar to this Issue : https://github.com/luc-github/ESP3D/issues/561
ESP32 is full, and ESP32 is full.
But there's no way the ESP32 has 1GB of storage.
Did I do something wrong?

code branch: 3.0
IDE: PlatformIO
env: esp32cam
board:AI Thinker esp32-cam
```
[env:esp32cam]
platform = https://github.com/platformio/platform-espressif32.git
;theboard
board = esp32dev
framework = arduino
monitor_speed = 115200
monitor_flags = --echo
monitor_filters = send_on_enter, colorize, esp32_exception_decoder
; set frequency to 240MHz
board_build.f_cpu = 240000000L
; set frequency to 80MHz
board_build.f_flash = 80000000L
board_build.flash_mode = qio
; None
build_flags =   -DCORE_DEBUG_LEVEL=0
                -DBOARD_HAS_PSRAM
                -mfix-esp32-psram-cache-issue
board_build.partitions = min_spiffs.csv
upload_speed = 921600
lib_ignore = 
    ESPAsyncTCP
```

[image]

luc-github, 2021-02-13:
the partition of FS seems not mounted  `/ does not exist` and the space is not 1023MB  (1GB ?) for min_spiffs, it means the Filesystem is corrupted

click on console you should get a terminal
type `[ESP710]FORMAT` wait for the message format...done then restart, you should have a proper Filesystem

hyzx86, 2021-02-13:
wow~ , it woked , Thanks a lot!

closed 2021-02-13 by hyzx86