luc-github/ESP3D #585: Won't compile in VSCode

Pugga12, 2021-02-18:
Compiling with VSCode doesn't work. Building for a ESP8266 1m

platformio.ini file:
`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
src_dir     = esp3d
build_dir   = .pioenvs
lib_dir     = libraries
libdeps_dir = .piolibdeps
data_dir = esp3d/data
default_envs = esp8266

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
;theboard
board = esp32dev
framework = arduino
monitor_speed = 115200
; set frequency to 240MHz
board_build.f_cpu = 240000000L
; set frequency to 80MHz
board_build.f_flash = 80000000L
board_build.flash_mode = qio
; None
build_flags = -DCORE_DEBUG_LEVEL=0
board_build.partitions = min_spiffs.csv
upload_speed = 921600
lib_ignore = 
    ESPAsyncTCP

[env:esp8266]
;esp8266 core 2.5.2
platform = espressif8266@2.2.0
platform_packages      = toolchain-xtensa@2.40802.200502
board = esp01_1m
framework = arduino
monitor_speed = 115200
; set frequency to 160MHz
board_build.f_cpu = 800000000L
; set frequency to 40MHz
board_build.f_flash = 40000000L
board_build.flash_mode = dio
upload_resetmethod = ck
board_build.ldscript = eagle.flash.1m256.ld
build_flags = 
    -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
    -DNONOSDK221=1
    -DNDEBUG
    -DVTABLES_IN_FLASH
upload_speed = 115200
lib_ignore = 
    AsyncTCP
    ESP32NetBIOS
    ESP32SSPD
`
Output 
`rocessing esp8266 (platform: espressif8266@2.2.0; board: esp01_1m; framework: arduino)
-------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 (2.2.0) > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 800MHz, 80KB RAM, 1MB Flash
PACKAGES:
 - framework-arduinoespressif8266 2.20502.0 (2.5.2)
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Converting esp3d.ino

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=0 to SConscript calls.
Missing SConscript 'C:\Users\adama\.platformio\packages\framework-arduinoespressif8266\tools\platformio-build.py'
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
File "C:\Users\adama\.platformio\platforms\espressif8266\builder\frameworks\arduino.py", line 33, in <module>
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESPAsyncTCP> 1.1.1
|-- <ESP Async WebServer> 1.2.0
|   |-- <ESPAsyncTCP> 1.1.1
|-- <DHT sensor library for ESPx> 1.0.6
|-- <WebSockets> 2.1.0
|   |-- <ESPAsyncTCP> 1.1.1
|-- <ESP8266_SSD1306> 3.2.7
Building in release mode
AttributeError: 'SConsEnvironment' object has no attribute 'ElfToBin':
  File "C:\Users\adama\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 177:       
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\adama\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\adama\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\adama\.platformio\platforms\espressif8266\builder\main.py", line 218:
    target_firm = env.ElfToBin(
===================================== [FAILED] Took 1.49 seconds =====================================

Environment    Status    Duration
-------------  --------  ------------
esp8266        FAILED    00:00:01.487
================================ 1 failed, 0 succeeded in 00:00:01.487 ================================The terminal process "C:\Users\adama\.platformio\penv\Scripts\pio.exe 'run', '--environment', 'esp8266'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
`

luc-github, 2021-02-18:
use latest git, also this issue looks Platformio package issue

Pugga12, 2021-02-19:
Apparently something got corrupted, re-installed PIO, and compiled

luc-github, 2021-02-19:
great have fun 😸

closed 2021-02-19 by luc-github