luc-github/ESP3D #590: [ESP600} issues - again!

Legsmaniac, 2021-02-24:
in your end code of cura just add 

*  `[ESP600] print done` if printing from SD
* `M118 [ESP600] print done` if streaming gcode from any host

_Originally posted by @luc-github in https://github.com/luc-github/ESP3D/issues/577#issuecomment-775264369_

I finally got around to trying this but it's still not working for me.
As you explained in the above issue, I added `[ESP600]You print is now complete` to the End gcode in Cura and I can see in the Terminal window of ESP3D is says in red  `echo:Unknown command: "[ESP600] Your print is complete"`   This is printing from file saved on the SD Card plugged into printer.

I also tried in PrusaSlicer but in order to get Prusa to accept the [ESP600] in the End gcode I apparently (according to Google research) have to type it as `{"ESP600"}` but that still doesn't work.

Any hints please?

luc-github, 2021-02-25:
try add M118 before

Legsmaniac, 2021-02-25:
I tried adding M118  and ESP3D Terminal reports, again in red `echo:Unknown command: "error: Wrong authentication!"`

Authentication?  Is this because I am using authentication in ESP3D admin and password set?
Again, this was using a file on SD Card in the printer, not "streaming from host" as you stated, which I assume means sending the gcode from PC via USB which I wouldn't be able to do since ESP3D is using the UART Serial wired internally.

Are these commands reported back in red sent my Marlin?  Does using Jyers modded Marlin bugfix-2.0.x have anything to do with it?

luc-github, 2021-02-25:
if you are using authencation yes you need to add password in command: 
https://github.com/luc-github/ESP3D/wiki/Direct-ESP3D-commands
```
* Send Notification
[ESP600]msg [pwd=<admin password>]
```

Legsmaniac, 2021-02-25:
OK, Thanks.
So I added [pwd=mypassword] to the end and double checked the password is correct, the result in Terminal is still the same, a red `echo:Unknown command: "error: Wrong authentication!"`  
Not really sure I'm impressed nor happy that my password has to be exposed in gcodes in full readable format.  Better if it was able to be a hashed version?  Tempted to turn off authentication which means no security exposing ESP3D through my router for outside access.  Unless I do this some other way.......
Sadly, there's no way in the settings of ESP3D to disable authentication which means I have to comment out in the config, recomplie and use the OTA update feature to reflash.  Maybe then my [ESP600} end gcode will work......

luc-github, 2021-02-25:
can you send me the line you send - just change the password changing letter by X but keep all others characteres so I can check if it is a bug in transmission

Well this is current limitation - I am thinking on allow to disable password for serial only,
 well hashed or not is still the same , the hash is visible and so can be used 

are you using v3.0 or 2.1 ?

Legsmaniac, 2021-02-25:
Sorry, it is working.  My bad.  Again.
I assumed you needed `[pwd=pass]`.  After I posted my last message, it got me wondering so I took the square brackets off so it now reads simply `pwd=pass` and Lo & Behold - an email arrived telling me my print was finished. 

You're right about the hash also being exposed, I can't believe I actually suggested that, LOL.
I was just thinking about this and what I posted in my last message while making a coffee.  Doh!  (Simpsons moment)

So, thank you for your help once again and sorry for my bad once again.  I can confirm it's working, in Cura at least.  Prusa is another story but that Prusa at fault, not ESP3D.  (Unless you can change the fact that you use square brackets in your commands as square brackets are used in Prusa as part of their command set.)

luc-github, 2021-02-25:
I used `[ ]` because no one used it and was good trigger... you mean prusa copy me ? rhooooo 
^_^ anywhay happy you make it work
I am checking that indeed serial may not need password as it is local access - I will fix this in 3.0, I did not think about all possible scenario yet

Legsmaniac, 2021-02-25:
LOL  Norty boy!

OK, Update......
It IS working in Prusa, just have to be careful in formatting the required code (and in MY case, careful with flamin' typos!!)
So the entry in Prusa I used was `M118 {"[ESP600]"}Your creation has completed! pwd=mypassword` and Prusa ignores square brackets as commands and takes them as (I presume) print entries.  Anyway, email received from Prusa too (I use both Cura and Prusa interchangeably, both have their pros and cons) so all is well with the world.

Yes, I'm currently using V3.0 by the way and yes, I'm currently local, obviously.  However, won't making it work on a local network without password rely on users having NAT Passthrough on their routers?  Not all routers support this, especially IP provided routers.  Ignore me if I'm talking cr.p again, I know nothing.  🤣  Just trying to help of course.

Anyway, all sorted again for now, many thanks for your patience with a silly old duffer like me.  😃

luc-github, 2021-02-25:
I am adding a flag to disable authentication only on serial by webUI , so it should cover:
* case when serial need to keep authentication 
* case people are confident on serial and so no need to put any password in GCODE

luc-github, 2021-02-25:
The setting to disable serial authentication is implemented https://github.com/luc-github/ESP3D/commit/eaba22990a156dc99713c94d95a580acd916620c
[image]

closed 2021-02-25 by luc-github

Legsmaniac, 2021-02-25:
Just to say thank you, working great, a very helpful modification.  😊