luc-github/ESP3D #591: [FEATURE REQUEST] Support for current Creality Silent boards V4.2.x

TzOk83, 2021-02-27:
Are there any plans for supporting the Creality Silent 32-bit boards, like v4.2.7?

luc-github, 2021-02-27:
what do you mean ?

TzOk83, 2021-02-27:
What I mean is mainly the Wiki Page for that board if it is possible to connect to it at all. #524 somehow answers my question, but official documentation/Wiki doesn't mention that boards at all, so I understand it is not tested/recommended.

luc-github, 2021-02-27:
I do not have all boards - board for wiki is done by users sharing their configuration - if not in wiki it just means no one did or share the configuration, it may not need it is not possible

Legsmaniac, 2021-02-27:
@TzOk83 I'm using Creality 4.2.7 with ESP3D v3.0 so if you're wanting to know if the 4.2.7 board works with ESP3D then yes, it does.

closed 2021-02-27 by luc-github