luc-github/ESP3D #593: Installation instructions for PlatformIO

salfter, 2021-03-01:
I'm not much of a fan of the Arduino IDE, preferring instead to use VSCodium and PlatformIO.  It took a bit of muddling around to get ESP3D installed on some ESP-01Ss I'd recently purchased (and yes, I even gave the Arduino IDE a shot, and failed), but I think I have it down to some fairly simple steps that you might consider including in the project documentation.

I'm assuming that you're using VSCodium (or Visual Studio Code) on Linux and that you have already installed PlatformIO within VSCodium.  If you're using a different editor or OS, adjust accordingly.  Steps 3 and 4 are likely to need the most adaptation for other environments, especially if you're running Windows.

1) You'll need to determine what flash mode is supported by your boards.  I've packaged up [CheckFlashConfig](https://gitlab.com/salfter/CheckFlashConfig) for use with PlatformIO.  Follow the instructions within and note what type (QIO, DIO, QOUT, DOUT) your boards use.
2) Check out [this repo](https://github.com/luc-github/ESP3D) and [the web interface repo](https://github.com/luc-github/ESP3D-WEBUI) to separate directories.
3) Go into ESP3D-WEBUI first.  Make sure you have NodeJS installed by whatever means is appropriate, then do ```sudo npm install --global gulp-cli && sudo npm install --global gulp@4.0.0 && npm install && gulp package``` to compile single-language and multi-language versions of index.html.gz.  The multi-language file will be in the root directory; the single-language files will be under languages.  
4) Since space is short on the ESP-01S, a single-language file is recommended.  Copy it to its place within the ESP3D repo: ```cp languages/en/index.html.gz ../ESP3D/esp3d/data/```
5) Open ESP3D in VSCodium.  Open platformio.ini, go down to the [env:esp01s_160mhz-2.7.4] section, find the board_build.flash_mode line, and change it to whatever type you got from step 1.
6) Open esp3d/config.h.  Commenting out ```#define NOTIFICATION_FEATURE``` is another option for keeping space free on the device, but it might be enough to just use a single-language web interface.   Uncommenting ```#define AUTHENTICATION_FEATURE``` would be a good idea for most people.  There are some other things to tweak within; read the comments.
7) Unplug your programmer (with attached ESP-01S), flip the switch on the programmer to "program," and plug it back in.
8) From the PlatformIO project tasks list, drill down through env:esp01s_160mhz-2.7.4 -> Platform and click "Erase Flash" to make sure the ESP-01S is cleared out.
9) Unplug your programmer and plug it in again.
10) From the PlatformIO project tasks list, drill down through env:esp01s_160mhz-2.7.4 -> Platform, click "Build Filesystem Image," allow it to complete, then click "Upload Filesystem Image" and wait for the upload to complete.
11) Unplug your programmer and plug it in again.
12) From the PlatformIO project tasks list, drill down through env:esp01s_160mhz-2.7.4 -> General and click "Upload."  Wait for the code to build and upload.
13) Unplug your programmer yet again, flip the switch to "UART," and plug it back in.

At this point, the software should be installed and is ready to be configured.  Go into the WiFi settings on your phone and wait for ESP3D to show up.  Connect to it (uses WPA, password is 12345678).  Android's captive-portal check will then take you to the initial setup for the device (possibly after a login screen; username and password are both "admin").  Follow the prompts to configure for your wireless network and your printer; when complete, unplug and replug the programmer one last time to let the device connect to your network, at which point it should be accessible to your (wired) computers and other devices.  You should also be able at this point to separate the ESP-01S from the programmer and connect it to your 3D printer's control electronics.  (The SKR Pro 1.1 in my Hypercube 300 has an 8-pin connector specifically for these gadgets.  Other boards will vary.)

luc-github, 2021-03-01:
thank you but you should not need to build webUI, I builld it when I commit usually
Did you meet issue with already built in version ?

salfter, 2021-03-01:
No issues with the bundled UI (I found it in the main package), but I substituted an English-only build for the multi-language build on the suggestion that it would save space (it's about a third smaller).

(At least I thought I'd seen it as a suggestion somewhere.  Of course, now I can't find where I thought I'd seen it. :) I might have a non-OTA-upgradable configuration; I haven't tried testing it yet.)

luc-github, 2021-03-01:
Great thank you - I am not good doing documentation ^_^
Doing PR on repository allows to easily add pictures when needed and to keep them in wiki, online allows easy edition but we lost a lot of pictures because they were online and so a lot of good information.

luc-github, 2021-03-01:
as you use VSCode there is a nice extension : Markdown Preview Enhanced

NateCroix, 2021-05-15:
How to go about setting a static IP for the client network address? I tried to set this up and the unit has become uncommunicative even after fully cleaning and reuploading both firmware and web environment. I'm not sure how to "improvise" the language stripping (steps 3+4) being on windows.

luc-github, 2021-05-15:
when board is connected to PC use the Monitor option,
then type [ESP444]RESET , then [ESP444]RESTART
the board should restart in AP mode 
you can also configure network using direct commands https://github.com/luc-github/ESP3D/wiki/Direct-ESP3D-commands

closed 2021-06-16 by stale[bot]