luc-github/ESP3D #594: [BUG] ESP3D doesn't connect to wifi

GrzesiuS, 2021-03-02:
**Describe the bug**
After the initial setup, ESP3D can't connect to wifi.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash new firmware
2. Setup ESP3D
3. Reboot the device

**Expected behavior**
Established connection to wifi

**Logs**
[13:56:57]M117 wifi
[13:56:58]M117 Connecting.   
[13:56:58]M117 Connecting..  
[13:56:59]M117 Connecting... 
[13:56:59]M117 Connecting....
[13:57:00]M117 Connecting.   
[13:57:00]M117 Connecting..  
[13:57:00]M117 Disconnected
[13:57:01]M117 Connecting... 
[13:57:01]M117 Connecting....
[13:57:02]M117 Connecting.   
[13:57:02]M117 Connecting..  
[13:57:03]M117 Connecting... 
[13:57:03]M117 Connecting....
[13:57:03]M117 Disconnected
[13:57:04]M117 Connecting.   
[13:57:04]M117 Connecting..  
[13:57:05]M117 Connecting... 
[13:57:05]M117 Connecting....
[13:57:06]M117 Connecting.   
[13:57:06]M117 Connecting..  
[13:57:06]M117 Disconnected
[13:57:07]M117 No SSID. 
[13:57:07]M117 No SSID..
[13:57:08]M117 No SSID.   
[13:57:08]M117 No SSID..  
[13:57:09]M117 No SSID... 
[13:57:09]M117 Disconnected
[13:57:09]M117 No SSID....
[13:57:10]M117 No SSID.   
[13:57:10]M117 No SSID..  
[13:57:11]M117 No SSID... 
[13:57:11]M117 No SSID....
[13:57:12]M117 No SSID.   
[13:57:12]M117 Disconnected
[13:57:12]M117 No SSID..  
[13:57:13]M117 No SSID... 
[13:57:13]M117 No SSID....
[13:57:14]M117 No SSID.   
[13:57:14]M117 No SSID..  
[13:57:15]M117 No SSID... 
[13:57:15]M117 Disconnected
[13:57:15]M117 No SSID....
[13:57:16]M117 No SSID.   
[13:57:16]M117 No SSID..  
[13:57:17]M117 No SSID... 
[13:57:17]M117 No SSID....
[13:57:18]M117 Not Connected!
[13:57:18]M117 Safe mode 1
[13:57:18]M117 wifi
[13:57:20]M117 192.168.0.1

**ESP3D Firmware:**
- Version: 2.1.1
- Wifi mode: STA]
- Flash method: PlatformIO
 
**Board used (please complete the following information):**
 - MCU: ESP12F
 - Flash size: 4M

luc-github, 2021-03-02:
it does not see the SSID properly- so my guess is signal is not strong enough 
Check the output of `[ESP420]` and `[ESP410]plain` and the strengh of the signal

Did you change anything in FW  or config file ?
the output `M117 wifi` is weird

GrzesiuS, 2021-03-02:
I tested it more and I have more observations. After I firmware flashing without clearing data, the device boots correctly the first time with settings from the previous setup, and it connects correctly to my network. After reboot, the device can't connect and show the same log I posted earlier.

I have 4 wifi points in my house. The wifi scanner shows 2 points have signal over 60%, the third one over 50%, so I assume signal strength is quite good.

I comment out NOTIFICATION_FEATURE as I saw on YT tutorial and changed ESP_DEFAULT_NAME to my printer name.

In plaftormio.ini I've changed the board type and CPU speed according to the tutorial.

[env:esp8266]
board = esp12e
board_build.f_cpu = 80000000L

That is all I changed in the project. Changes look harmless IMO.

I tried to send the commands you suggested and after reboot, the device isn't connected to the network but when I open a port like this:

$port= new-Object System.IO.Ports.SerialPort COM5,115200,None,8,one
$port.open()

I see a led blink and the device is connecting to Wi-fi with success and works fine.

luc-github, 2021-03-02:
you must keep 160Mz as set in PIO, 80Mz is too slow
use webUI in AP mode to show the output of commands I have requested - what external wifi scanner show is not necessarly what esp itself see - that is why commands need to be done from ESP

GrzesiuS, 2021-03-02:
The problem is when the device says:

[20:18:34]M117 Not Connected!
[20:18:34]M117 Safe mode 1
[20:18:34]M117 wifi
[20:18:37]M117 192.168.0.1

I assume it works in AP mode but I can't find ESP3D network like it doesn't broadcast SSID. Even if I used SSID saved in the phone I can't connect. The phone says a network is out of range.

I can execute commands in STA mode when I use powershell trick described above:

[ESP420]plain
Chip ID: 11206345
CPU Frequency: 160Mhz
Free memory: 38.47 KB
SDK: 2.2.1(cfd48f3)
Flash Size: 4.00 MB
Available Size for update: 604.20 KB(Ok)
Available Size for SPIFFS: 1.86 MB
Baud rate: 115200
Sleep mode: Modem
Channel: 11
Phy Mode: 11g
Web port: 80
Data port: 8888
Hostname: Ender3Pro
Active Mode: STA (48:3F:DA:AA:FE:C9)
Connected to: wifi
Signal: 68%
IP Mode: DHCP
IP: 192.168.1.16
Gateway: 192.168.1.1
Mask: 255.255.255.0
DNS: 192.168.1.2
Disabled Mode: AP (4A:3F:DA:AA:FE:C9)
Captive portal: Enabled
SSDP: Enabled
NetBios: Disabled
mDNS: Enabled
Web Update: Enabled
Pin Recovery: Disabled
Authentication: Disabled
Target Firmware: Marlin
M117 output: Enabled
Serial output: Enabled
Web socket  output: Enabled
TCP output: Enabled
FW version: 2.1.1.b8 ESP8266/8586

[ESP410]plain
wifi	64	Secure
wifi	30	Secure
wifi	78	Secure
wifi	50	Secure

luc-github, 2021-03-02:
I am not sure how ESP SDK will behave when have 4 AP with same SSID - I am guessing it will try to connect to the one he get answer first, if it is the one with 30%, it may after loose the connection as signal is low and so failed to connect, if it connect to another one with enough signal it succeed. This part is not handled by ESP3D code but espressif SDK.

The-Font, 2021-03-11:
I had the issue that it connected en then tried to connect again but after a timeout it put itself in AP. If I had it connected to my phone as a hotspot no problems but connecting to a Fritbox Router would not work. I found a youtube video that described that for an ESP8266 with 1M memory you need to disable some things because the file is to big.

config.h

Disable notification feature
//NOTIFICATION_FEATURE : allow to push notifications
//#define NOTIFICATION_FEATURE

After that I got it up and running and even connecting to a Fritzbox.

The youtube video I was talking about: https://www.youtube.com/watch?v=NGgzw-XayEo

Legsmaniac, 2021-03-11:
> for an ESP8266 with 1M memory

I would have thought that the better solution in this case would be to use a better board.  They're cheap enough after all.  😊

luc-github, 2021-03-11:
@The-Font I think there is misunderstanding - disabling Notifications feature is not a memory issue, it is flash size issue  for web update - The origin of this advice is on wiki : https://github.com/luc-github/ESP3D/discussions/331

As you mention you can connect to your phone as hotspot - connection to another one does not need more memory it is exactly same.

Also the difference between ESP12 nodemcu and ESP01 is not memory but only flash size: 4M vs 1M

I do not know what the issue is, but if it works by disabling Notifications (which is not even started when doing connection) then I am stumped...

The-Font, 2021-03-11:
It is a flash size issue you are right not a memory issue. I was using the esp01 because that one fits nice on my TFT35 display. What is a good/better alternative? ESP12 or ESP32 and is connecting this to the TFT35 display from bigtreetech possible?

luc-github, 2021-03-11:
if you have ESP01 it is ok for the TFT35 - the only issue is to do web update due to the 1M flash - but you not need to update ESP3D so often

Use a 4M version is a good alternative - use a nodemcu or a reworked   ESP01 should be ok - but using a nodemcu or wemos take space when esp01 is designed for TFT

ESP32 is more powerful but also currently still little bit slower than the ESP8266 due to the very agressive core code optimisation in ESP8266
Instead of use another board I tried the ESP01 rework like mentioned on https://github.com/luc-github/ESP3D/discussions/331 and was not easy for my fat fingers and old eyes ^_^

Because what is nice with ESP01 is the small form factor, so I got this idea : https://github.com/luc-github/ESP3D/discussions/531
If you like challenge you can give a try - I had no time to test it yet  😸

The-Font, 2021-03-12:
Thanks for all the work and effort you put into this project. I don't have the skills to rework an esp01, same issue as you mention ;)
I might look into a board that has more memory installed already and figure out how to fit it. But for now I keep trying the ESP01. For me it is a very nice extra but not needed, I also have a raspberry with octopi connected to the printer.

luc-github, 2021-03-15:
that is nice finding :grin:

luc-github, 2021-03-16:
but looking at screen it use :
– W25Q80: 8M-bit / 1M -byte (1,048,576) 

So I suspect the 4MB is a mistake

The-Font, 2021-03-17:
I did see the foto on their page also and contacted them. They confirmed it was a 4M. I got one here now and yes it is a 4M. The memory chip is different then on their foto.

The one I received from them showed:

#esptool.exe flash_id
esptool.py v3.0
Found 1 serial ports
Serial port COM15
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...

The 1M I already had shows:

#esptool.exe flash_id
esptool.py v3.0
Found 1 serial ports
Serial port COM15
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Manufacturer: 5e
Device: 6014
Detected flash size: 1MB
Hard resetting via RTS pin...

See also the photo:

[image]

luc-github, 2021-03-17:
great then - that is good news 👍

luc-github, 2021-04-02:
@GrzesiuS is issue still valid ?

GrzesiuS, 2021-04-02:
I have found a reason. It isn't working when ESP12 is on the programming board.

After proper wiring, it's working fine.

closed 2021-04-02 by GrzesiuS

luc-github, 2021-04-02:
Ok thank you