luc-github/ESP3D #598: [Question]ESP works on USB adapter but not on SKR1.4

alequeiroz, 2021-03-11:
Hello, been trying to set a ESP01 on SKR 1.4 for a few days, using latest marlin, when i flash ESP3D on usb adapter it displays wifi and serial output is right, i can connect and set everything up, but when i plugged to SKR, it dosent even blink the blue led for a breif moment, wifi (AP mode, first time connecting) dosen´t show up, i followed Chris Riley tutorial (latest Marlin firmware there are some thing already done), uploaded via VS and also arduino IDE, but the same error persists, M43 from marlin shows the rigth pinouts, baudrate is equal both in Marlin and ESP3D, but the weirdest thing is when i connect Serial adapter to SKR Serial 3 (ESP place) it connects correctly, voltage is rigth on this adaptar as well, but when i place ESP on SKR nothing happens.
Any sugestions? 
PS* i´ll edit this if needed formating or more infos, just ask.
Thanks!

luc-github, 2021-03-11:
the esp3d is not a BTT one but from another place - I am correct ?

alequeiroz, 2021-03-11:
[image]
Yes, it´s a common one, like this

luc-github, 2021-03-11:
that is the problem, you need to solder CH-PD to 3.3v , SKR 1.4 do not do it for you , BTT ESP01 are self soldered so there is no issue: check https://github.com/luc-github/ESP3D/wiki/ESP8266-01

I raised issue and solved : https://github.com/luc-github/ESP3D/issues/516#issuecomment-708944840

alequeiroz, 2021-03-11:
Worked like a charm, thank you very much! :D

closed 2021-03-11 by alequeiroz

luc-github, 2021-03-11:
Great have fun then ^_^