luc-github/ESP3D #604: ESP01S module does not generate WIFI hotpot

JosemiZubi, 2021-03-29:
After generating the firmware and uploading it to the module, I look for the network generated by the ESP01s module but it does not exist. I'm not doing something basic but I cannot know what I am doing wrong.

ESP3D Firmware: 2.1.1 ESP3D for ESP8266
- ESP01s
- Flash method: PlatformIO
    
Thanks in advance!!!

[image]

JosemiZubi, 2021-03-30:
Hi:

Using Arduino IDE for generating the firmware, the module works, it's OK

Thanks.

luc-github, 2021-03-30:
you are not using platformio to flash , screenshot show another application

JosemiZubi, 2021-03-30:
Hi, Thank you very much for the reply.

  Then, I must also use platformio to flash. How should i do it?: NodeMCU ESP8266, Arduino extensión...What is't the best way?. Thanks on avance.

luc-github, 2021-03-30:
you just connect your board select the esp8266 and click upload and monitor
com detection is automatic and all settings are ready
[image]

I have just updated th platformio.ini as I have noticed the flash method was not using the nodemcu one so you may need to update lastest git

JosemiZubi, 2021-03-30:
Hi:

O.K. I will use thids method. Thanks you for your support and your grear job!

luc-github, 2021-03-31:
is it ok now ? if yes please close issue

JosemiZubi, 2021-03-31:
It'so o.k. I clóse the issue. Thanks.

closed 2021-03-31 by luc-github