luc-github/ESP3D #607: [BUG] T0 and T1 temperatures are not displayed independently

Drzet, 2021-04-02:
**Describe the bug**
While printing with only 1 extruder ESP3D displays the same temperature for both working and idle extruder  

**Expected behavior**
Correct temperature for every tool

**Screenshots**
[image]

**ESP3D Firmware:**
- Version:2.1.1b8
- Web UI Version: 2.1b57

**Target Firmware:**
 - Name: Marlin
 - Version 2.0.x.bugfix

**Board used (please complete the following information):**
SKR v1.4

luc-github, 2021-04-02:
The latest webUI is 2.1b69  you are using  2.1b57 
Can you confirm issue is still present in latest version ?

Drzet, 2021-04-02:
I've uploaded the latest index.html.gz, now it shows webUI version 2.1.b60
Issue still present

luc-github, 2021-04-02:
latest webUI is 2.1b69  in git not 2.1.b60 and I think issue was solved in 2.1b61 - please use latest git

Drzet, 2021-04-02:
They had b60 within latest release .zip
Got b69, issue still present
[image]

luc-github, 2021-04-02:
ok I will have a look

Drzet, 2021-04-02:
Seems like it's reporting the "T", not separately T0..Tx

luc-github, 2021-04-02:
Yes indeed - I am on it, I think I fixed this in 3.0 - I will fix this one too

luc-github, 2021-04-02:
[index.html.gz](https://github.com/luc-github/ESP3D/files/6250084/index.html.gz)
Please let me know if this one fix the issue -

Drzet, 2021-04-02:
[image]

luc-github, 2021-04-02:
ok great I push the update on github - thank you for reporting issue and feedback

luc-github, 2021-04-02:
WebUI updated - I move issue to webUI