luc-github/ESP3D #608: Ok I effectively had a bad wiring / A10 GT2560 Geeetech

petervflocke, 2021-04-03:
@Arthur75012 in https://github.com/luc-github/ESP3D/issues/457#issuecomment-624222855_
mentioned that he successfully wired GT2560, can some one provide the wiring?
Was it connected to J7 Uart interface with RXD3/TXD3 or into the connection between uC and U3/CH430 using RXD0/TXD0?

Any suggestion is very much welcome!

Arthur75012, 2021-04-05:
Hello
This is my wiring : I had to weld 2 wires on theses tow resistors (see pictures) to get the rx/tx connection
[image]

petervflocke, 2021-04-05:
Ok, so you use the  connection between uC and U3/CH430 using RXD0/TXD0.
OK, many thanks for responding.

closed 2021-04-05 by petervflocke