luc-github/ESP3D #609: [Question]MKS-Wifi and pure Marlin

brdvvoz, 2021-04-05:
I have Marlin+MKS-Wifi+ESP3D combo and I really want to not be constrained by MKS version of Marlin, but I can't get ESP3D and Marlin to speak - Marlin tab in webui gives me 404 "Cannot get valid data". 
How can I debug/fix this? Don't really know where to start.

luc-github, 2021-04-05:
what is your FW versions ?

brdvvoz, 2021-04-05:
ESP3D is 3.0.0.a94
Marlin 2.0.7.2
Robin Nano 1.2 and MKS Robin Wifi 1.0

luc-github, 2021-04-05:
is only Marlin Tab ? or all commands that do not work?

brdvvoz, 2021-04-05:
Commands don't work - I see movement, temperature etc. interfaces, but no temperature readings from printer and all actions are ignored

luc-github, 2021-04-05:
did you set MKS_SERIAL https://github.com/luc-github/ESP3D/blob/3.0/esp3d/configuration.h#L43

`#define COMMUNICATION_PROTOCOL MKS_SERIAL `

brdvvoz, 2021-04-05:
Yes, MKS_Serial in config and 
Also baudrate is 250000 in both Marlin and webui

luc-github, 2021-04-05:
no for MKS baudrate must be 115200 this is MKS design

brdvvoz, 2021-04-05:
No, still doesn't work
Weirdly enough It kinda looks like mks wifi code is somehow connected to lvgl which I don't use
I'll try  compiling Marlin with lvgl and see what happens

luc-github, 2021-04-05:
to be honest I never tested original Marlin as MKS Marlin version has several fixes not present in mainstream - I have also even raised several bugs to MKS but did not test since a while

brdvvoz, 2021-04-05:
It seems that original Marlin has most of the related MKS code and I managed to compile fw with this code successfully. It had some effect (page loading way faster and it now shows "Loading... 2B" before fail)
BUT it still fails in the same way in the end
BUT I flashed MKS Marlin as sanity check and it behaves exactly the same so maybe there is something wrong in my ESP3D build after all? For instance, there are board parameters in Arduino IDE not listed in installation manual - Crystal Frequency, EspressifFW, Port etc. - which I left on default

brdvvoz, 2021-04-05:
[image]

This is the board config - am I missing something?

luc-github, 2021-04-05:
pls choose 2MB FS 1019 OTA because like this web update will not work, but it is not your issue currently
for other looks good

What `[ES420]plain` and `M115` give as output ?

brdvvoz, 2021-04-05:
M115 returns `ok
FIRMWARE_NAME:Robin_nano`
[ES420]plain returns nothing, though I'm not sure what this is
M503 just returns "Ok"
Also movement actually works now, but temperature monitoring doesn't

luc-github, 2021-04-05:
ok so communication is working
I wrote [ESP420]plain not [ESP420], so it is not recognized

for temperatures you need to enable poilling commands and add M105
I remember M503 is filtered by MKS code it is one thing I have raised to them - so for that it is a bug in MKS code not ESP3D

brdvvoz, 2021-04-05:
Polling  works and esp420 returns `chip id: 3281
CPU Freq: 160 Mhz
free mem: 23.27 KB
SDK: 2.2.2-dev(38a443e)
flash size: 4.00 MB
size for update: 476.55 KB
FS type: LittleFS
FS usage: 120.00 KB/2.98 MB
baud: 115200
sleep mode: none
wifi: ON
hostname: esp3d
HTTP port: 80
Telnet port: 23
WebDav port: 8181
sta: ON
mac: C8:2B:96:2B:0C:D1
SSID: Cat's Cradle
signal: 0 %
phy mode: 11n
channel: 13
ip mode: dhcp
ip: 192.168.1.184
gw: 192.168.1.1
msk: 255.255.255.0
DNS: 192.168.1.1
ap: OFF
mac: CA:2B:96:2B:0C:D1
serial: ON
notification: OFF
Serial: MKS
Target Fw: marlin
FW ver: 3.0.0.a94
FW arch: ESP82XX `

I will now try to decouple mkswifi from lvgl and get it working on normal Marlin, Thanks a lot for the help! (and unexpectedly immediate reply).

luc-github, 2021-04-05:
you can disable WebDav port: 8181 it is useless as it does not access to SD 
I see also 
Cat's Cradle signal: 0 %

Your signal is may be almost too low so you may have some connection issues

closed 2021-04-06 by luc-github