luc-github/ESP3D #611: [BUG] MKS Robin Nano 1.2 reboots on file upload to SD

BaltasarParreira, 2021-04-06:
Hi, I'm experience this problem every time I try to upload a file to the printer SD card.
The uploading message shoes on the MKS TFT35 but them screen pauses for a while and the board/printer reboots showing the MKS logo and beeps like if it was just powered on.

I'm running the latest Mks-Robin-Nano-Marlin2.0-Firmware that I just compiled and using the #define MKS_WIFI_MODULE.

Is this really a bug? Is there a solution?
Thanks.

luc-github, 2021-04-06:
as bot mentionned please fill information requested in template

BaltasarParreira, 2021-04-06:
Ok, I just done that on first post.

luc-github, 2021-04-06:
I do not see the version of the Robin Firmware - I have no clue of what is the problem without knowing on what the problem happen
Especially when it is the printer FW which restart - not ESP3D

Additionnally using NodeMCU V3 is bad idea, you should stick to MKS WIFI module as upload use very high speed baude rate that does not support any EMI noise, which will corrupt communication,  which will likely happen if you use another ESP board connected with wires

BaltasarParreira, 2021-04-06:
I have a couple of nodemcu's lying around so I was testing before I spend money buying the real MKS WIFI module.
Don't knew that EMI noise could interfere like you said, I never had problems with a lot of other things that I use them,

The  M115 command just gives this:
```
ok
FIRMWARE_NAME:Robin_nano
```

Is there any other that I can test/use?

luc-github, 2021-04-06:
EMI noise are not critical when using a baudrate at 115200 like commands  use, but upload use 
`#define UPLOAD_BAUD_RATE    1958400` so it become very sensitive to serial data corruption as MKS protocol does not use checksum neither retry on failure
anyway the board which is restarting is your Robin so issue is in Marlin not ESP3D as it was working until now but I definitly think using NodeMCU with MKS protocol is bad idea

about  the Marlin Tab and Robin also a Marlin MKS issue - https://github.com/luc-github/ESP3D/discussions/610#discussioncomment-574000

BaltasarParreira, 2021-04-06:
Ok I see, can I lower that baudrate like changing that setting to other lower value and compile?
Do I need to do some changes also on Marlin side, like on Configuration.h or Configuration_adv.h?
I recall seeing there the possibility to declare port speeds.
Or could it be possible to use other MKS Robin Nano serial port instead the reserved to connect the WIFI module?

luc-github, 2021-04-06:
in Marlin you need to change the speed in MKS code https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/blob/master/Marlin/src/lcd/extui/lib/mks_ui/wifiSerial.h#L33

The MKS wifi port is not managed by Configuration.h or Configuration_adv.h

BaltasarParreira, 2021-04-06:
Yes I know the extra 2 pins, but connecting those everything strangely stops working that give me thinking what could be the behavior of ESP3D with the real MKS WIFI module since all the pins are wired.
I don't recall seeing any setting to tell what type of ESP8266 we are using on ESP3D. right?

luc-github, 2021-04-06:
>Yes I know the extra 2 pins, but connecting those everything strangely stops working that give me thinking what could be the behavior of ESP3D with the real MKS WIFI module since all the pins are wired.

I do no not understand sorry

>I don't recall seeing any setting to tell what type of ESP8266 we are using on ESP3D. right?

what do you mean ?

BaltasarParreira, 2021-04-06:
For my test with the Nodemcu I just use the TX and RX pins apart the power (GND + 3.3V) since if I connect the D3 and D2 (supposed the extra 2 pins) nothing works.
But since I'm connecting with wires I can do that.
The real MKS WIFI module has all ESP8266 pins connected so they are inserted on the board socket. 
So unless I'm using some wrong pins for those 2 extra pins the MKS WIFI module will not work as well since it has them connected. Did you understand?

luc-github, 2021-04-06:
The 2 extra pins are necessary and defined:
https://github.com/luc-github/ESP3D/blob/3.0/esp3d/src/modules/mks/mks_service.cpp#L33-L35

BaltasarParreira, 2021-04-06:
But ESP3D use them as well in this case?

luc-github, 2021-04-06:
I just linked you esp3d FW code....of course, they are mandatory https://github.com/makerbase-mks/MKS-WIFI

luc-github, 2021-04-06:
what is you [`ESP420]plain` output ?

BaltasarParreira, 2021-04-06:
I get this:
> [ESP420]plain
chip id: 11547
CPU Freq: 160 Mhz
free mem: 23.68 KB
SDK: 2.2.1(cfd48f3)
flash size: 4.00 MB
size for update: 1023.00 KB
FS type: LittleFS
FS usage: 120.00 KB/1.98 MB
baud: 115200
sleep mode: none
wifi: ON
hostname: esp3d
HTTP port: 80
Telnet port: 23
sta: ON
mac: 84:F3:EB:EE:2D:1B
SSID: Balta-WIFI
signal: 88 %
phy mode: 11n
channel: 6
ip mode: dhcp
ip: 192.168.0.20
gw: 192.168.0.254
msk: 255.255.255.0
DNS: 192.168.0.254
ap: OFF
mac: 86:F3:EB:EE:2D:1B
serial: ON
notification: OFF
Serial: MKS
Target Fw: marlin
FW ver: 3.0.0.a94
FW arch: ESP82XX

luc-github, 2021-04-06:
Ok so ESP3D settings are correct
do if your wiring is correct sending `M105` or `M115` should also work

BaltasarParreira, 2021-04-06:
> M105
ok
T:25.7 /0.0 B:22.8 /0.0 T0:25.7 /0.0 T1:0.0 /0.0 @:0 B@:0

> M115
ok
FIRMWARE_NAME:Robin_nano

luc-github, 2021-04-06:
Upload works in exactly same way but using baud rate 1958400  instead of 115200, but rely more on pins GPIO0 and GPIO4 for notification and ack as it use several packets

BaltasarParreira, 2021-04-06:
Ok going to recheck those pins them, I think I was using other since on the MKS schematic for the module they are IO1 and IO2 that I have to check on the nodemcu mapping.
I let you know later after more tests with those wired too.
Thanks for the support, all the help and this amazing project !!!

luc-github, 2021-04-06:
IO0 and IO2 do not refer to ESP8266 GPIO but MKS internal naming

BaltasarParreira, 2021-04-06:
Ok using the module pinout:
[image]

and the MKS Robin Nano board:
[image]

I can see that they map to D0 and D4 that you showed on:
https://github.com/luc-github/ESP3D/blob/3.0/esp3d/src/modules/mks/mks_service.cpp#L33-L35

BaltasarParreira, 2021-04-06:
Solved, uploads are working without any problems.
I used the following pin mapping:

   Arduino IDE code  |    Marked on Nodemcu
---------------------------------------------------
              0                 |        GPIO 0 or D3
---------------------------------------------------
              4                 |        GPIO 4 or D2
---------------------------------------------------

Now only the Marlin Tab and Robin also a Marlin MKS issue - #610 (comment) is not working, all the rest is ok.

ANyway, what is that tab supose to do? Get some Marlin information only?

luc-github, 2021-04-06:
it show Marlin EEPROM settings:

[image]

BaltasarParreira, 2021-04-06:
Ok I see, so that is not working here.
 I found now that temperatures even polling enabled don't show nothing.
Also I switch the bed and I can't see it, no fields or something for the bed:
[image]

[image]

luc-github, 2021-04-06:
What poilling command did you put ?

BaltasarParreira, 2021-04-06:
I just switched this on since was off:
[image]

Does it need some sort of tuning or changing the those values?

luc-github, 2021-04-06:
you add the one you need - M105 for temps - M114 for posistions  - etc...
the answer will be seen in monitor

BaltasarParreira, 2021-04-06:
Perfect is showing now, I had to reboot the printer or the nodemcu, since cutting the power does that on both.
Anyway after a few time the printer beeps and reboots the board, after coming back the nodemcu looses connection with the board and everything stopped working.
Only recycling power makes it working again.
Without the polling on, this doesn't happen.

luc-github, 2021-04-06:
Sending Serial command should not make the printer beep - may be some bug in Marlin FW since I did the porting - may be the serial buffer is overflowed ?
You better ask MKS I think

luc-github, 2021-04-08:
@BaltasarParreira any update ?

BaltasarParreira, 2021-04-08:
Hi, no I'm stuck and can't find why or which command makes the printer reboot once in a while, so printing with the nodemcu connected is not an option for now as stops the current printing because of constant reboots.

luc-github, 2021-04-08:
does disable poiling solve the issue ?

BaltasarParreira, 2021-04-08:
Yes it solves if not doing nothing, but sending a file to print gives the same problem once in a while.
There is not really a specific time or fixed interval for the reboots.

luc-github, 2021-04-08:
so seems a buffer overflow on serial on printer board - did not have such issue, I have a Robin Nano 3.5 v2

BaltasarParreira, 2021-04-08:
I'm waiting to arrive a real ESP-12S like the original MKS WIFI module, them I'll make further tests because even using a nodemcu with the MKS WIFI firmware I get the same reboots, so it could be something related of the nodemcu because of the additional electronics it has. The MKS WIFI module is basic simple only having the ESP and nothing else, no pull up or down resistors here and there, etc.

luc-github, 2021-04-08:
I close issue then - feel free to open new one if any

closed 2021-04-08 by luc-github

BaltasarParreira, 2021-04-08:
Ok just some news and I think I solved the reboot problem.

I had compiled and installed on the printer the MKS firmware from this source:
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware
This was giving all the problems I reported.

So I decided to search and go for a more recent one and I got the latest from Marlin Bugfix Branch that could be not 100% tested and safe but has a lot of fixes from here:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x
I compiled with all the exact same settings and for now I have pooling on did a couple of files upload and no reboots.

The old get information on Marlin tab still exists behaving the same but the rest I think is normal.
Going to do more extended tests and prints and so on.

One other thing, why ESP3D doesn't support long file names and always changes the upload filename to something like old DOS in size 8 chars dot 3 chars for the extension?
The MKS Robin Nano SD supports long file names, there is even a setting for that on the Configuration(*).h files.

BaltasarParreira, 2021-04-08:
A new thing that now I have and not before is a second Extrusion temperature icon showing and also on the Extrusion box.
Off course since my printer only has one the values for this second are always empty.

Don't you think that should be good on settings a way of saying if we have one or two?

luc-github, 2021-04-08:
>One other thing, why ESP3D doesn't support long file names and always changes the upload filename to something like old DOS in size 8 chars dot 3 chars for the extension?

ESP3D do not do anything in 3.0 - it is MKS Marlin code which do that - Marlin does not support long name -  https://github.com/luc-github/ESP3D/discussions/326

It may be  bug in code, the number of extruders are automaticaly  detected based on M105 output
What Marlin respond when you type `M105`  ?

BaltasarParreira, 2021-04-08:
I got this:

[image]

On the Configuration(x).h files I have everywhere only one Extruder set.

luc-github, 2021-04-08:
Yes MKS output is buggy
I have fixed it before with MKS Marlin version when I tested :
https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/src/components/printer/temperatures.js#L254-L259

It looks like Marlin and MKS Marlin have different  code again...orz

I will add support for "0" in addition to "0.00" and provide you a version later

luc-github, 2021-04-08:
here the fix for buggy second extruder :
[index.html.gz](https://github.com/luc-github/ESP3D/files/6280984/index.html.gz)

The emergency stop send M112 https://marlinfw.org/docs/gcode/M112.html
You may need to enable emergency parser to make it faster but not all board have it supported

BaltasarParreira, 2021-04-08:
Ok thanks, I'll test it right away.

Regarding the M112 so the behaviour is as suppose to be reading from that Marlin link.
But here don't you think that a normal stop printing button doing the same function just for stop like the one in the front printer panel will be good?

luc-github, 2021-04-08:
the sd pause and sd stop button are not yet implemented in 3.0 https://github.com/luc-github/ESP3D-WEBUI/discussions/94#discussioncomment-192037

luc-github, 2021-04-08:
great 😸

BaltasarParreira, 2021-04-08:
Another suggestion, what about a confirmation message yes or no like the one from uploading a file to the print icon in front of each file?
Like this we don't accidentally fire up some prints that either to stop them only with the emergency stop that sort of halt the printer or going in front of her and do a manual stop. Booth option to solve we are to get up and go to the printer. ; )

luc-github, 2021-04-08:
Yes will be done, I am on rewrite of webUI 3.0 and I planned to add it, it easier now to add confirmation box

BaltasarParreira, 2021-04-09:
Update:
After all the upload file tests and navigation on the ESP3D interface without reboots I went for real printing tests and now the problem is printing from the files or even using the normal MKS WIFI module with Cura using the MKS plugin the printer becomes like crazy moving all around here and there in all X, Y and Z but nothing really prints in the end.
So no normal operation.

2º Update:
The crazy printing problem is everywhere, even normal printing from SD card does the same, so I conclude that this version is no good and apart fixing the rebooting problem can't be used for printing anything.

luc-github, 2021-04-09:
I have only used MKS source code when I ported ESP3D in december 2020 you can give a try of the code from this time

[image]

BaltasarParreira, 2021-04-09:
Maybe I give it a try from an older release, like the one highlighted here:
[image]

Tomorrow I'll check that, but a read and search a lot here an there and looks like that these crashes, reboots related with WIF are a WATCHDOG problem.
I switched it off and the upload works but in the end freezes without doing nothing.

Legsmaniac, 2021-04-09:
> Maybe I give it a try from an older release

Or maybe get a proper board.  🤣

I was considering the MKS but after reading several reviews and posts on forums by others who have tried/still have MKS boards which were far from favourable, I decided against it.  And it's looking like you're another one with unfavourable problems so case rested I guess.

BaltasarParreira, 2021-04-09:
There is no problem with the board, is a great 32bit board with a lot of options and the stock firmware works perfect and also the normal Marlin 2.0 with they're UI.
The problem here is wanted to get rid of the SD in/out work to print, there is not many boards supporting WIFI.
Of course I could use a octopi or some similar approach, but why having more things around if internally the board supports this.
The old stock firmware works 100%, but we want always what we can't have.
If it's red everybody wants green and vice versa... ; )

Legsmaniac, 2021-04-09:
Good that it works OK for you, Bud.  I'm only going by what I've heard from several friends and acquaintances on forums etc. and reviews.

I tried out OctoPrint, rather tentatively because I have never liked RPi's and sure enough, I didn't like Octo either.  RPi's take way too long to boot up.  Not so with ESP.  
By the way, I got around the SD file storage problem and 8.3 filename problem by using FlashAir WiFi SD card.  I can send files wirelessly to the card from my printer and thru ESP3D too, using a Panel I set up.  The SD stays permanently in the printer now.  😉

BaltasarParreira, 2021-04-09:
Yes that is another option I though, but since implies buying the WIFI SD, first I thought of trying using what I already have around, like a lot of nodemcu, raspberry pi's and so on.
From all of this I should get a perfect and good solution for my needs without extra costs.

luc-github, 2021-04-09:
@Legsmaniac you should do a github with your extra panels ans share them - they also could be promoted here https://github.com/luc-github/ESP3D/wiki#nice-things-done-using-esp3d

Legsmaniac, 2021-04-09:
I thought about doing that as, like you, I have loadza bitza kit.  I was going to make my own Arduino WiFi SD but then I saw an offer for a FlashAir real cheap so I got one.  It was a neater and tidy option because to use yet another ESP meant more wiring, more power to supply it and another box to make for it or somewhere to locate it.  I just bought an extension microSD to SD and printed a holder for it that slots into the V Rails.  Nice n neat n tidy.

Legsmaniac, 2021-04-09:
@luc-github OK, I could do. Although three of them are just URL links to things - one for HomeAssistant, one for my camera and one for the WiFi SD, they're nothing technical really.  

[image]

The only one I have created using HTML (so far) is one for my LED lighting but again, it's nothing complicated and only good for anyone who might be using WLED like I am.

[image]

luc-github, 2021-04-09:
@Legsmaniac my meaning is : you did nice job, be proud of it 😉 
I am sure lot of people will benefit of your job

Legsmaniac, 2021-04-09:
> you did nice job, be proud of it 😉

Awwww, shucks, thank you kind sir.  😊
But it wouldn't have been possible without you and ESP3D so you, too did a nice, no, excellent job, be proud of it.  😉

BaltasarParreira, 2021-04-10:
New update:
Ok for now I abandoned the MKS UI firmware and MKS WIFI, so I compiled the normal latest Marlin 2.0 Bugfix Branch and going to test this using normal RAW_SERIAL protocol since the MKS Robin Nano 1.2 without the WIFI module uses that as a serial port.

So far success, everything is working, the pooling and even the Marlin information tab as you can see;
[image]

Only problem is that the upload files button disappeared so I can't send files to print them as you can see here:
[image]

Weird... did I messed up some configuration on settings or there is a bug for the 3-0 version using normal protocol?

Waiting for you comments and help. ; )

[21 more replies]