luc-github/ESP3D #616: M486 Inclusion

Legsmaniac, 2021-04-15:
**Is your feature request related to a problem? Please describe.**
Not a problem except currently it shows up in Terminal as Unknown Command, obviously because it is not implemented,. 

**Describe the solution you'd like**
I'd like to see the inclusion, if possible of M486 verbose comments, same as OctoPrint gives.  Currently, it displays in red Unknown Command so it it's receiving the "command", wouldn't it be possible to interpret the command and display the comment inserted by PrusaSlicer? 

**Describe alternatives you've considered**
None, not interested in RPi.

**Additional context**

[image]

https://marlinfw.org/docs/gcode/M486.html

https://github.com/paukstelis/PrusaSlicer-M486

luc-github, 2021-04-15:
What about to put `Unknow command` instead it allow to cover more commands ?

luc-github, 2021-04-15:
Additional question : why using the M486 if it is not supported by your Marlin configuration ? this looks like a post processing implementation or I misunderstood how it was generated ?

Legsmaniac, 2021-04-15:
Whilst M486 may not be interpreted by Marlin, it's intended to be interpreted by OctoPi.  This means that any object comments given by Prusa triggered by the M486 instruction would be received and actioned by the RPi running Octo and therefore appear on the screen of the RPi.  I wondered if the M486 instructions could be intercepted by ESP3D and displayed in the Terminal panel instead of Unknown Command and possibly the LCD display screens if any used on ESP3D devices.

luc-github, 2021-04-15:
the Unknow command is made by Marlin, not ESP3D
Currently I try to not include any GCODE filtering because it may slow down the bridge, it may change when esp8266 will not be supported anymore.

Also be noted filtering won't work if you print from SD. 

As it is post GCODE post processing slicer opperation, why not add another post processing one to the slicer that change `M486`  to `M117`  or to  `[ESP214]` if ESP3D has display ?

Legsmaniac, 2021-04-15:
> why not add another post processing one to the slicer that change `M486` to `M117` 

Now why didn't you think of that?!?!

🤣👍

closed 2021-04-15 by Legsmaniac

Legsmaniac, 2021-04-20:
***  UPDATE  ***

Further to this......
I discovered that all I had to do was uncomment` #define CANCEL_OBJECTS` in the `Configuration_adv.h` file and recompile Marlin.
#define CANCEL_OBJECTS was commented out by default.

I did this and reflashed the printer, now I see the comments information created by Prusa and SuperSlicer directly in the Terminal panel window and none of the errors.

Thought I'd add this for the benefit of others in case anyone else has the same query.  😉