luc-github/ESP3D #628: Error after updateing MKS wi-fi esp12e[Question]<Enter comprehensive title>

themacboy, 2021-05-06:
Good morning,

I just compiled Esp3d to install in my Mks ribin wi-fi module, and all looks right, but after uploading the index.html.gz i get this error:

[image]

Compiling word fine, my configurationis (and I know it worked for other people):
[image]

[image]

Not sure what im doing wrong or how to check it

luc-github, 2021-05-06:
I need FW version and webUI version you are using

themacboy, 2021-05-06:
I have downloaded v.3 version:
https://github.com/luc-github/ESP3D/tree/3.0

and webUI this file, form 2.1:

https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz

No sure if there is a v.3 html

luc-github, 2021-05-06:
so this is your problem - the read me is pretty clear:
[image]

And the first boot embedded page menu also link to webUI 3.0

themacboy, 2021-05-06:
lol sorry. I will test in next 3 hours and close it

themacboy, 2021-05-06:
Solved, your were right.

I picky needed additional help from other user, due filesystem wasn't removed after update and error continues. Other member told me to tribal original Mks bin to reset all the chip, and later reinstall Esp3d, and woeked fibre. 

Thx all

closed 2021-05-06 by themacboy

luc-github, 2021-05-07:
use the embedded  maintenance page
`htttp://yourip?forcefallback=yes`

themacboy, 2021-05-07:
> use the embedded maintenance page
> `htttp://yourip?forcefallback=yes`

Thx, nice to know

themacboy, 2021-05-11:
> @themacboy <https://github.com/themacboy> what request?

LoL wrong github jajaja sorry